dart-archive/sdk #305: Cleanup documentation generation for fletch packages

ricowind, 2015-11-11:
@karlklose @sgjesse 
I will land a change that makes us archive documentation for our packages.

The documentation will be put into the sdk bundle and be available online.
There is a number of hacks in the current implementation:
  The current implementation is linux only (we only have a forked version for linux on gcs)
  The way the libraries files is used is hackish, Karl et al is working on making that better
  This is currently mixed up with the sdk bundling, we may want to completely separate it
  We should try to convince dartdoc people that one should be able to give dartdoc a set of packages that should be documented, the output should include an overall page with links to the individual packages

mit-mit, 2015-11-16:
I booked a meeting with a few dartdoc folks to discuss this.

mit-mit, 2015-12-07:
The docs for packages work fine now. We still need to figure out what to do for core libs. Maybe try to reexport them like we do for packages?

mit-mit, 2016-07-18:
Closing, @sigurdm fixed this in https://github.com/dartino/sdk/commit/acf9af6231e46e4df34fd7d14a6c1ab0c8d75766

closed 2016-07-18 by mit-mit