dart-archive/sdk #308: Linker errors iOS

racarone, 2015-11-12:
When building the github sample, compilation succeeds but when trying to run the example in Xcode, it gives linker errors about undefined symbols (from the public API) when trying to link libfletch.

zerny, 2015-11-12:
Hi, the github sample project is very experimental, that said we would like to resolve any build issues it might have. I have not been able to reproduce the problem here so it would be helpful with some more info from your end.

Did you follow the setup described in [samples/github/ios/compile.sh](../tree/master/samples/github/ios/compile.sh)? Namely:

Install and build fletch and cocapods;
Run samples/github/ios/compile.sh;
Run pod install;
Load github.workspace in Xcode and run it.

If this fails, please attach specifics of the failure, your platform, target device and xcode version and I'll see if I can reproduce the issue.

racarone, 2015-11-12:
Yeah I built fletch, ran compile.sh, pod install, opened the workspace and ran it. I'm on OSX El Capitan, running Xcode 7.1 target iOS simulator and device with 9.1 SDK.

zerny, 2015-11-12:
Thanks. Could you also paste the error log here. In particular the linking command and the reported failures?

racarone, 2015-11-12:
Ok I did a git reset and pull in order to start fresh. New errors. Here are my steps and output after a glcient sync.

Compile fletch:
```
nop:fletch racarone$ ninja -C out/DebugIA32Clang
ninja: Entering directory `out/DebugIA32Clang'
[47/182] SOLINK libffi_test_library.dylib, POSTBUILDS
ld: warning: directory not found for option '-L../../third_party/libs/mac/x86'
[52/182] SOLINK libmdns_extension_lib.dylib, POSTBUILDS
ld: warning: directory not found for option '-L../../third_party/libs/mac/x86'
[109/182] LINK natives_to_json, POSTBUILDS
ld: warning: directory not found for option '-L../../third_party/libs/mac/x86'
[126/182] LINK fletch_vm_library_generator, POSTBUILDS
ld: warning: directory not found for option '-L../../third_party/libs/mac/x86'
[161/182] LINK fletch, POSTBUILDS
ld: warning: directory not found for option '-L../../third_party/libs/mac/x86'
[161/182] LINK fletch_for_sdk, POSTBUILDS
ld: warning: directory not found for option '-L../../third_party/libs/mac/x86'
[165/182] ACTION libfletch: generate_l...letch_b0b3e2c42820e005912164f0add07903
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/libtool: file: ../../out/DebugIA32Clang/libfletch_vm_library_base.a(fletch_vm_library_base.event_handler_linux.o) has no symbols
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/libtool: file: ../../out/DebugIA32Clang/libfletch_vm_library_base.a(fletch_vm_library_base.event_handler_lk.o) has no symbols
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/libtool: file: ../../out/DebugIA32Clang/libfletch_vm_library_base.a(fletch_vm_library_base.event_handler_cmsis.o) has no symbols
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/libtool: file: ../../out/DebugIA32Clang/libfletch_vm_library_base.a(fletch_vm_library_base.ffi_disabled.o) has no symbols
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/libtool: file: ../../out/DebugIA32Clang/libfletch_vm_library_base.a(fletch_vm_library_base.ffi_static.o) has no symbols
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/libtool: file: ../../out/DebugIA32Clang/libfletch_vm_library_base.a(fletch_vm_library_base.ffi_linux.o) has no symbols
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/libtool: file: ../../out/DebugIA32Clang/libfletch_vm_library_base.a(fletch_vm_library_base.native_process_disabled.o) has no symbols
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/libtool: file: ../../out/DebugIA32Clang/libfletch_vm_library_base.a(fletch_vm_library_base.natives_lk.o) has no symbols
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/libtool: file: ../../out/DebugIA32Clang/libfletch_vm_library_base.a(fletch_vm_library_base.natives_cmsis.o) has no symbols
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/libtool: file: ../../out/DebugIA32Clang/libfletch_vm_library_base.a(fletch_vm_library_base.thread_lk.o) has no symbols
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/libtool: file: ../../out/DebugIA32Clang/libfletch_vm_library_base.a(fletch_vm_library_base.thread_cmsis.o) has no symbols
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/libtool: file: ../../out/DebugIA32Clang/libfletch_shared.a(fletch_shared.native_socket_linux.o) has no symbols
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/libtool: file: ../../out/DebugIA32Clang/libfletch_shared.a(fletch_shared.native_socket_lk.o) has no symbols
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/libtool: file: ../../out/DebugIA32Clang/libfletch_shared.a(fletch_shared.platform_linux.o) has no symbols
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/libtool: file: ../../out/DebugIA32Clang/libfletch_shared.a(fletch_shared.platform_lk.o) has no symbols
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/libtool: file: ../../out/DebugIA32Clang/libfletch_shared.a(fletch_shared.platform_cmsis.o) has no symbols
[179/182] LINK shared_cc_tests, POSTBUILDS
ld: warning: directory not found for option '-L../../third_party/libs/mac/x86'
[180/182] LINK flashtool, POSTBUILDS
ld: warning: directory not found for option '-L../../third_party/libs/mac/x86'
ld: warning: PIE disabled. Absolute addressing (perhaps -mdynamic-no-pic) not allowed in code signed PIE, but used in _InterpretFast from libfletch_vm_library.a(fletch_vm_library.generated.o). To fix this warning, don't compile with -mdynamic-no-pic or link with -Wl,-no_pie
[180/182] LINK fletch-vm, POSTBUILDS
ld: warning: directory not found for option '-L../../third_party/libs/mac/x86'
ld: warning: PIE disabled. Absolute addressing (perhaps -mdynamic-no-pic) not allowed in code signed PIE, but used in _InterpretFast from libfletch_vm_library.a(fletch_vm_library.generated.o). To fix this warning, don't compile with -mdynamic-no-pic or link with -Wl,-no_pie
[181/182] LINK vm_cc_tests, POSTBUILDS
ld: warning: directory not found for option '-L../../third_party/libs/mac/x86'
[... cut ...]
+ rm -rf /Users/racarone/Github/fletch-repo/fletch/samples/github/generated/service
+ mkdir -p /Users/racarone/Github/fletch-repo/fletch/samples/github/generated/service
+ /Users/racarone/Github/fletch-repo/fletch/out/ReleaseIA32/fletch x-servicec file /Users/racarone/Github/fletch-repo/fletch/samples/github/generated/immi/idl/immi_service.idl out /Users/racarone/Github/fletch-repo/fletch/samples/github/generated/service
'/Users/racarone/Github/fletch-repo/fletch/out/ReleaseIA32/fletch' isn't a supported action. Try running 'fletch help'
```

peter-ahe-google, 2015-11-12:
@racarone I think you need to run `killall dart fletch fletch-vm`.
Also, you may need to run `pub get` from `tools/immic`.

racarone, 2015-11-12:
Ah you were correct about the killall. Back to the linker errors:

```
Ld /Users/racarone/Library/Developer/Xcode/DerivedData/github-ceqszkwdgmvuorbyivhbfobgoklk/Build/Products/Debug-iphonesimulator/github.app/github normal i386
    cd /Users/racarone/Github/fletch-repo/fletch/samples/github/ios
    export IPHONEOS_DEPLOYMENT_TARGET=8.1
    export PATH="/Applications/Xcode.app/Contents/Developer/Platforms/iPhoneSimulator.platform/Developer/usr/bin:/Applications/Xcode.app/Contents/Developer/usr/bin:/usr/local/bin:/usr/bin:/bin:/usr/sbin:/sbin"
    /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/clang++ -arch i386 -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/iPhoneSimulator.platform/Developer/SDKs/iPhoneSimulator9.1.sdk -L/Users/racarone/Library/Developer/Xcode/DerivedData/github-ceqszkwdgmvuorbyivhbfobgoklk/Build/Products/Debug-iphonesimulator -L/Users/racarone/Github/fletch-repo/fletch/samples/github/ios -F/Users/racarone/Library/Developer/Xcode/DerivedData/github-ceqszkwdgmvuorbyivhbfobgoklk/Build/Products/Debug-iphonesimulator -filelist /Users/racarone/Library/Developer/Xcode/DerivedData/github-ceqszkwdgmvuorbyivhbfobgoklk/Build/Intermediates/github.build/Debug-iphonesimulator/github.build/Objects-normal/i386/github.LinkFileList -Xlinker -rpath -Xlinker @executable_path/Frameworks -mios-simulator-version-min=8.1 -Xlinker -objc_abi_version -Xlinker 2 -v -ObjC -lFletch -lImmi -lImmiSamples -lLruCache -lMMDrawerController -lService -framework QuartzCore -stdlib=libc++ -fobjc-arc -fobjc-link-runtime -lPods -lfletch -Xlinker -dependency_info -Xlinker /Users/racarone/Library/Developer/Xcode/DerivedData/github-ceqszkwdgmvuorbyivhbfobgoklk/Build/Intermediates/github.build/Debug-iphonesimulator/github.build/Objects-normal/i386/github_dependency_info.dat -o /Users/racarone/Library/Developer/Xcode/DerivedData/github-ceqszkwdgmvuorbyivhbfobgoklk/Build/Products/Debug-iphonesimulator/github.app/github

Apple LLVM version 7.0.0 (clang-700.1.76)
Target: i386-apple-darwin15.0.0
Thread model: posix
 "/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/ld" -demangle -dynamic -arch i386 -ios_simulator_version_min 8.1.0 -syslibroot /Applications/Xcode.app/Contents/Developer/Platforms/iPhoneSimulator.platform/Developer/SDKs/iPhoneSimulator9.1.sdk -ObjC -o /Users/racarone/Library/Developer/Xcode/DerivedData/github-ceqszkwdgmvuorbyivhbfobgoklk/Build/Products/Debug-iphonesimulator/github.app/github -L/Users/racarone/Library/Developer/Xcode/DerivedData/github-ceqszkwdgmvuorbyivhbfobgoklk/Build/Products/Debug-iphonesimulator -L/Users/racarone/Github/fletch-repo/fletch/samples/github/ios -filelist /Users/racarone/Library/Developer/Xcode/DerivedData/github-ceqszkwdgmvuorbyivhbfobgoklk/Build/Intermediates/github.build/Debug-iphonesimulator/github.build/Objects-normal/i386/github.LinkFileList -rpath @executable_path/Frameworks -objc_abi_version 2 -lFletch -lImmi -lImmiSamples -lLruCache -lMMDrawerController -lService -framework QuartzCore -lPods -lfletch -dependency_info /Users/racarone/Library/Developer/Xcode/DerivedData/github-ceqszkwdgmvuorbyivhbfobgoklk/Build/Intermediates/github.build/Debug-iphonesimulator/github.build/Objects-normal/i386/github_dependency_info.dat -force_load /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/lib/arc/libarclite_iphonesimulator.a -framework Foundation -lobjc -lc++ -lSystem /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../lib/clang/7.0.0/lib/darwin/libclang_rt.ios.a -F/Users/racarone/Library/Developer/Xcode/DerivedData/github-ceqszkwdgmvuorbyivhbfobgoklk/Build/Products/Debug-iphonesimulator
Undefined symbols for architecture i386:
  "_FletchDeleteProgram", referenced from:
      +[AppDelegate loadAndRunDartSnapshot] in AppDelegate.o
  "_FletchLoadSnapshot", referenced from:
      +[AppDelegate loadSnapshot:] in AppDelegate.o
  "_FletchRunMultipleMain", referenced from:
      +[AppDelegate loadAndRunDartSnapshot] in AppDelegate.o
  "_FletchSetup", referenced from:
      -[AppDelegate application:didFinishLaunchingWithOptions:] in AppDelegate.o
  "_FletchTearDown", referenced from:
      -[AppDelegate applicationWillTerminate:] in AppDelegate.o
  "_FletchWaitForDebuggerConnection", referenced from:
      +[AppDelegate loadAndRunDartSnapshot] in AppDelegate.o
  "_ServiceApiInvoke", referenced from:
      Builder::InvokeMethod(void*, void*) in libService.a(struct.o)
      GithubMockServer::start(int) in libService.a(github_mock.o)
      GithubMockServer::stop() in libService.a(github_mock.o)
      ImmiServiceLayer::reset(unsigned short) in libService.a(immi_service.o)
      ImmiServiceLayer::refresh(unsigned short) in libService.a(immi_service.o)
      ImmiServiceLayer::dispatchVoid(unsigned short) in libService.a(immi_service.o)
      ImmiServiceLayer::dispatchInt32Int32(unsigned short, int, int) in libService.a(immi_service.o)
      ...
  "_ServiceApiInvokeAsync", referenced from:
      Builder::InvokeMethodAsync(void*, void*, void (*)(void*), void*, void*) in libService.a(struct.o)
      GithubMockServer::startAsync(int, void (*)(void*), void*) in libService.a(github_mock.o)
      GithubMockServer::stopAsync(void (*)(void*), void*) in libService.a(github_mock.o)
      ImmiServiceLayer::resetAsync(unsigned short, void (*)(void*), void*) in libService.a(immi_service.o)
[... cut ...]
  "_ServiceApiSetup", referenced from:
      -[AppDelegate application:didFinishLaunchingWithOptions:] in AppDelegate.o
  "_ServiceApiTearDown", referenced from:
      -[AppDelegate applicationWillTerminate:] in AppDelegate.o
  "_ServiceApiTerminate", referenced from:
      GithubMockServer::tearDown() in libService.a(github_mock.o)
      ImmiServiceLayer::tearDown() in libService.a(immi_service.o)
ld: symbol(s) not found for architecture i386
clang: error: linker command failed with exit code 1 (use -v to see invocation)
```

peter-ahe-google, 2015-11-12:
I think @zerny has signed off for the day, so you'll have to do with my guesses :-)

I don't have experience with pod files, but the linker command doesn't seem to include `-L/Users/racarone/Github/fletch-repo/fletch/out/ReleaseIA32`. So I'd try adding this symlink:

```
ln -s /Users/racarone/Github/fletch-repo/fletch/out/ReleaseIA32/libfletch.a /Users/racarone/Github/fletch-repo/fletch/samples/github/ios
```

racarone, 2015-11-12:
Well there's already a `libfletch.a` in that directory that's composed of the `i386` and `armv7` archs (confirmed with `lipo`), it's definitely there. It's trying to link it at ` -lfletch` I believe. I can start a blank project in Xcode and link to the `libfletch.a` static library, but if I call any of the above functions it results in the same linker errors.

zerny, 2015-11-13:
FYI, Xcode 7 introduced some changes in linking and now requires that libraries specifically for OSX or iOS even though the library is the same in the case of running it on the simulator (the current setup works for Xcode 6). A fix for this issue requires some changes to our build system which I have started work on.

zerny, 2015-11-13:
A fix for this issue is uploaded at https://codereview.chromium.org/1446433003

The quick fix is to rename libfletch.a to libfletchvm.a (or something else) from within Xcode. The main issue was that libfletch.a conflicts with the cocoapod Fletch and thus fails to be loaded when linking.

closed 2015-11-13 by zerny

racarone, 2015-11-13:
Great works perfectly now, thank you!