dart-archive/sdk #309: SD card flashing program nits

lukechurch, 2015-11-12:
```
lukechurch-macbookpro:fletch-sdk lukechurch$ platforms/raspberry-pi2/flash_sd_card 
Password:
This program will prepare an SD card for the Raspberry Pi 2
Please remove all SD cards from this computer, and then press enter
Please insert the SD card you want to use with Fletch, and then press enter
Found the SD card /dev/rdisk3. Press enter to use this card. This will delete all contents on the card.
Enter the name of the device - default is 'fletch': 
Enter the static IP address you would like the device to have. Leave it blank to have it assigned via DHCP:192.168.1.111
Downloading SD card image
Downloading: DONE                             
Decompressing SD card image.
Unmounting the SD card.
Writing image to SD card.
Writing: 1136656384 bytes transferred in 127.071268 secs (8945031 bytes/sec)
```

Improvements:

- Is it possible to have the program state what it's going to do before requesting an admin password?
- In this case, please terminate lines with a "."
- press enter should be "press Enter."
- It would be great if we could show a bit more information about the card than the /dev/rdisk3. Can we list the name of the drive and the root of the file system?
- If a user wishes to cancel wiping their card, how should they do that? 
- For default is 'fletch': consider adding (press Enter to accept)
- For download the writing, it would be helpful to use friendly representations of the numbers, e.g. the equivalent of ls -lh rather than ls -l.

sgjesse, 2015-11-16:
CL https://codereview.chromium.org/1444353002 addresses the issues like this:

- _Is it possible to have the program state what it's going to do before requesting an admin password?_
Yes, the two scripts `flash-sd-card` and `configure-network` now both prints `This program requires superuser to run.`

- _In this case, please terminate lines with a "."_
Done.

- _press enter should be "press Enter."_
Done

- _It would be great if we could show a bit more information about the card than the /dev/rdisk3. Can we list the name of the drive and the root of the file system?_

Added information on the card.

On Linux it will show something like this:
```
Please insert the SD card you want to use with Fletch. Then press Enter.

Found the SD card /dev/sdd.

Disk /dev/sdd: 31.7 GB, 31724666880 bytes
64 heads, 32 sectors/track, 30255 cylinders, total 61962240 sectors
Units = sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disk identifier: 0xba2edfb9

   Device Boot      Start         End      Blocks   Id  System
/dev/sdd1            8192      122879       57344    c  W95 FAT32 (LBA)
/dev/sdd2          122880     8447999     4162560   83  Linux

This will delete all contents on the card. Press Enter to use this card (Ctrl-C to stop).
```

On Mac OS it will show something like this:
```
Please insert the SD card you want to use with Fletch. Then press Enter.

Found the SD card /dev/rdisk2.

/dev/disk2
   #:                       TYPE NAME                    SIZE       IDENTIFIER
   0:     FDisk_partition_scheme                        *31.7 GB    disk2
   1:             Windows_FAT_32 boot                    58.7 MB    disk2s1
   2:                      Linux                         4.3 GB     disk2s2

This will delete all contents on the card. Press Enter to use this card (Ctrl-C to stop).
```

- _If a user wishes to cancel wiping their card, how should they do that?_
Press Ctrl-C. This is now stated - see bullet above.

- _For default is 'fletch': consider adding (press Enter to accept)_
Done.

- _For download the writing, it would be helpful to use friendly representations of the numbers, e.g. the equivalent of ls -lh rather than ls -l._
Downloading now display like this:
Downloading: received 58 Mb of 1291 Mb
I did not change the messages for writing the card, as that would require parsing the output from dd.

sgjesse, 2015-11-17:
Fixed in https://github.com/dart-lang/fletch/commit/7db59ead89cd6ff2a1a4984a9de4b0362685ff95.

closed 2015-11-17 by sgjesse