dart-archive/sdk #310: Confusing fletch quit interaction

lukechurch, 2015-11-12:
```
lukechurch-macbookpro:fletch-sdk lukechurch$ ./bin/fletch --version
0.1.0-dev.6.0
lukechurch-macbookpro:fletch-sdk lukechurch$ ./bin/fletch quit
Background process wasn't running
lukechurch-macbookpro:fletch-sdk lukechurch$ ./bin/fletch --version
0.2.0-dev.3.0
```

This was during an upgrade.
As fletch quit reported "Nothing was running for me to quit", it's unclear why the reported version changed on re-execution.

sgjesse, 2015-11-12:
Can this be caused by a compiler from before the version was communicated from the CLI?

wibling, 2015-11-12:
I suspect this is due to the way we detect which persistent process is
running. Peter mentioned something about the background process socket file
changing when upgrading meaning the new sdk would not find the old process.
I am not sure but I will try to repro and see if it makes sense.

peter-ahe-google, 2015-11-12:
This is what happens:

1. We use lsof to find processes using the Dart VM binary bundled in the SDK. There are no such processes.

2. We don't kill any process, but still delete the socket file. The persistent process is still running, but nobody can contact it.

3. Next time you run `fletch ...` a new persistent process is started. This process creates a new socket file.

wibling, 2015-11-12:
What Peter said ;) I seem to be able to reproduce this, so I will look into
using pgrep instead of lsof to determine whether an existing persistent
process is running.

/gustav

closed 2015-11-16 by wibling