dart-archive/sdk #311: Review the upgrading of the agent on a device (including the x-upgrade command)

sgjesse, 2015-11-12:
When the fletch-agent on a remote device (currently always Raspberry Pi 2) has a different version of the SDK on the host the following message is displayed:

```
$ fletch run samples/raspberry_pi/basic/blinky.dart in session remote
Could not start vm on device because the compiler and the session's remote device have different versions.
Compiler version: '0.2.0-dev.3.0'
Device version: '0.2.0-dev.2.0'.
Try running 'fletch x-upgrade agent with file <agent debian package> in session remote'
```

One suggestion would be to replace <agent debian package> with the actual path to use.

The current x-upgrade command looks like this:

```
$ fletch x-upgrade agent with file platforms/raspberry-pi2/fletch-agent_0.2.0-dev.3.0-1_armhf.deb in session remote
```

When missing some of he words in this command an error like this is displayed:

```
bin/fletch x-upgrade platforms/raspberry-pi2/fletch-agent_0.2.0-dev.3.0-1_armhf.deb 
The Fletch program is broken and has crashed.

The Fletch team would greatly appreciate if you would take a moment to
report this problem at https://github.com/dart-lang/fletch/issues/new

Please include the following information:

* the name and version of your operating system

* the Fletch SDK version (0.2.0-dev.3.0)

* the entire message you see here (including the full stack trace below)

Error when formatting diagnostic:
  kind: DiagnosticKind.verbRequiresTargetButGot
  template: Can't perform '#{verb}' without '#{requiredTarget}', but got: '#{target}'
  arguments: {#{verb}: Verb("x-upgrade"), #{target}: NamedTarget(TargetKind.FILE, "platforms/raspberry-pi2/fletch-agent_0.2.0-dev.3.0-1_armhf.deb")}
  missingParameters: #{requiredTarget}
  unusedParameters: 
  formattedMessage: Can't perform 'x-upgrade' without '#{requiredTarget}', but got: 'NamedTarget(TargetKind.FILE, "platforms/raspberry-pi2/fletch-agent_0.2.0-dev.3.0-1_armhf.deb")'
#0      Diagnostic.formatMessage (package:fletchc/src/diagnostic.dart:141:7)
#1      ClientController.reportErrorToClient (package:fletchc/src/driver/driver_main.dart:474:44)
#2      _RootZone.runBinary (dart:async/zone.dart:1171)
#3      _Future._propagateToListeners.handleError (dart:async/future_impl.dart:521)
#4      _Future._propagateToListeners (dart:async/future_impl.dart:580)
#5      _Future._propagateToListeners.handleWhenCompleteCallback.<anonymous closure> (dart:async/future_impl.dart:560)
#6      _RootZone.runUnary (dart:async/zone.dart:1166)
#7      _Future._propagateToListeners.handleValueCallback (dart:async/future_impl.dart:494)
#8      _Future._propagateToListeners (dart:async/future_impl.dart:577)
#9      _Future._addListener.<anonymous closure> (dart:async/future_impl.dart:274)
#10     _microtaskLoop (dart:async/schedule_microtask.dart:43)
#11     _microtaskLoopEntry (dart:async/schedule_microtask.dart:52)
#12     _Timer._runTimers (dart:isolate-patch/timer_impl.dart:394)
#13     _Timer._handleMessage (dart:isolate-patch/timer_impl.dart:414)
#14     _RawReceivePortImpl._handleMessage (dart:isolate-patch/isolate_patch.dart:148)
```

sgjesse, 2015-11-12:
/cc @karlklose

lukechurch, 2015-11-13:
I agree we should replace the fix text with the actual command to run so it can be copied and pasted.

We could probably also use some work on the command as well. It's really long.

Could we make some parts of it implicit? I.e. if you're in the SDK and there is one file called fletch-agent_blah_.deb we could use that.

The use of 'file' should also be optional.

karlklose, 2015-11-24:
The following issues have been addressed:
- 0179f1a: x-upgrade works without giving a file; it looks for a package in platforms/<platform>/ and lets the user provide a file name if it found no package.
- ce256e2, 550a995 address crashes when formatting messages
- bc42ec1: in case of a version conflict, look for the package with matching version in platforms and suggest to just use 'x-upgrade agent in session <session>' if it is there. Otherwise it suggests to get the matching SDK first.

karlklose, 2015-11-27:
Reassigning to @lukechurch for UX review.

lukechurch, 2015-12-11:
LGTM!

closed 2015-12-11 by lukechurch