dart-archive/sdk #312: LK tests on qemu get confused when tests time out

madsager, 2015-11-12:
There are a couple of tests that sometimes timeout while writing a stack overflow exception stack-trace on qemu. The script is not smart enough to figure out that this is a timeout and reports it as a MissingRuntimeError. We should fix that.

I'll mark the tests as slow to give them more time on the bots to finish printing.

Here is an example:

```
FAILED: fletchc-fletchvm debug_ia32 language/issue_22780_test/01
Expected: Pass 
Actual: MissingRuntimeError
CommandOutput[fletch_session]:

stdout:
Previous commands in this session:

$ out/DebugIA32/fletch create session 3 with .fletch-settings

 => 0
$ out/DebugIA32/fletch show log in session 3
[3: 2015-11-12 01:46:33.919 created session with file:///mnt/data/b/build/slave/fletch-lk-debug-arm-qemu/build/fletch/.fletch-settings Settings(packages: file:///mnt/data/b/build/slave/fletch-lk-debug-arm-qemu/build/fletch/.packages, options: [], constants: {}, device_address: null, device_type: DeviceType.mobile)]

 => 0
$ out/DebugIA32/fletch-vm &
$ out/DebugIA32/fletch attach tcp_socket 127.0.0.1:44099 in session 3

 => 0
$ out/DebugIA32/fletch export /mnt/data/b/build/slave/fletch-lk-debug-arm-qemu/build/fletch/out/DebugIA32/generated_tests/language/issue_22780_test_01.dart to file /mnt/data/b/build/slave/fletch-lk-debug-arm-qemu/build/fletch/out/DebugIA32/generated_compilations/fletchc/multitest_language_issue_22780_test_01/fletch.snapshot in session 3

 => 0
$ out/DebugIA32/fletch x-end session 3
Ended session '3'.

 => 0
$ out/DebugIA32/fletch-vm

stderr:
Fletch VM exitcode is 0

CommandOutput[vm]:

stdout:
Starting qemu...
Started with PID 32506
Waiting for qemu to come up...
QEMU 2.4.0.1 monitor - type 'help' for more information
(qemu) Starting fletch...
Waiting for size...
Sending size (30885)...
Waiting for snapshot request...
Sending snapshot...
..............................
SNAPSHOT READ.
STARTING fletch-vm...
LOADING snapshot...
RUNNING program...
Uncaught exception:
a Instance: 
  - class = class

Frame  0: Function(3564)
Frame  1: Function(3564)
Frame  2: Function(3564)
...
Frame  10150: Function(3564)
Frame  10151: Function(3564)
Frame  10152: FunctionTEARING DOWN fletch-vm...
EXIT CODE: 255
Killing 32506
] 
Killing 32506

stderr:
qemu: terminating on signal 15 from pid 32500
tools/lk/run_snapshot_lk_qemu.sh: line 84: /tmp/tmp.on8BA8xxgv/qemu.out: No such file or directory
tools/lk/run_snapshot_lk_qemu.sh: line 44: kill: (32506) - No such process
```

madsager, 2015-11-12:
@skabet

skabet, 2015-11-24:
I've identified the issue as data being dropped somewhere between printf in LK, and the output to the named pipe, from qemu. Will investigate further.

skabet, 2015-11-27:
Fixed with new version of qemu.

closed 2015-11-27 by skabet