dart-archive/sdk #314: Review the x-discover command

sgjesse, 2015-11-12:
The `x-discover` command is a helper to check the connected devices (right now it can report Raspberry Pi's). Here is the output with two Pi's connected:

```
$ bin/fletch x-discover
Looking for Fletch capable devices...
Found device fletch.local on address 192.168.1.199.
Found device fletch-2.local on address 192.168.1.111.
```

How to promote this to a non `x-` command?

New name `list-devices`?

Something completely different?

sgjesse, 2015-11-12:
/cc @mit-mit

kasperl, 2015-11-12:
$ fletch show devices
?

sgjesse, 2015-11-12:
Looking at `show_verb.dart`, then `show` is required to have a session, and for just shoving the connected devices we don't want to force creating a session.

@peter-ahe-google, @wibling Is there a way to have a verb which in some cases requires a session and sometimes not (e.g. `show log` vs. `show devices`?

peter-ahe-google, 2015-11-12:
You should always run in a session and use its worker isolate. Especially true for anything that runs a long process that may need to be aborted.

sgjesse, 2015-11-12:
Just listing connected devices is a "general" command that should be independent of any active session. If that disables Ctrl-C that is of course unfortunate (that is the case for `x-discover` as it is implemented now). 

Unless we have a way of running in a transient temporary session, It doesn't make sense to run `show devices` in a session.

lukechurch, 2015-11-13:
Lets separate the UI from the implementation here.

I think ``` show devices ``` is probably the right UI. I guess we need to adjust the implementation?

sgjesse, 2015-11-13:
As `show devices` is session agnostic it can run in any session. If there are no sessions the local session will be created automatically, and it will run in there. One can then terminate the discovery using Ctrl-C with the side effect that whatever session it is running in gets terminated.

A CL for that behaviour is here: https://codereview.chromium.org/1438163003.

lukechurch, 2015-11-13:
Could we also reformat the text a little to:

```
Fletch capable devices:
Device at 192.168.0.1    fletch-thing.local
Device at 192.168.0.111  fletch-thing.local
```

(So that we can get tab alignment to support users visually scanning the list)

mit-mit, 2015-11-13:
Implemented in 0.2

closed 2015-11-13 by mit-mit