dart-archive/sdk #316: Fletch Deploy command

mit-mit, 2015-11-13:
```fletch run``` is great during development, but we don't have a good way of making a program run after you are done writing it. This tracks adding a new command, which models how traditional embedded devices are developed:

```
fletch deploy <file name> in session <session name>
```

This should compile and snapshot <file name>, copy it to the device attached in <session name>, and start it. Rebooting the device should cause the program to be restarted.

To stop a device from running anything, or to just remove the code from it you can run:

```
fletch deploy erase in session <session name>
```

mit-mit, 2015-11-13:
@lukechurch, please review

wibling, 2015-11-16:
We talked a bit about how this could work. One suggestion is to have 'fletch build' command which creates a Debian package containing a snapshot, the fletch-vm, and some setup for starting the program on boot.
The package can be installed manually or if the fletch agent is installed on a device a separate command like 'fletch install <path to package>' could be used to install the package via the agent.
A separate 'fletch uninstall <package name>' could be to uninstall the package via the agent.

This approach has the benefit of leverage the Debian package system and it is also option to have the fletch-agent part of the device, ie. the user can choose to manually install the package instead of using the fletch install command.