dart-archive/sdk #320: Running our tests seems to use "fletch show log"

mkustermann, 2015-11-16:
When running our tests it seems like we are running "fletch show log" for each test. Since the log gets bigger and bigger and `tools/test.py` accumulates stdio from commands for all tests, the memory usage of `./tools/test.py` can grow to **beyond 1GB**.
=> This is an issue on our small ARM boards.

For normal commands, there is support for pruning stdio - see [here](https://github.com/dart-lang/fletch/blob/master/tools/testing/dart/fletch_session_command.dart). But tests with a session use a different infrastructure, namely the one in [tools/testing/dart/fletch_session_command.dart](https://github.com/dart-lang/fletch/blob/master/tools/testing/dart/fletch_session_command.dart).

There are different ways to mitigate this, for example:
  * Do not run 'fletch show log' for all tests
  * Let the 'fletch show log' show only the last X commands and not everything
  * Prune the log data when reading output from 'fletch show log'
  * Try to not keep all stdio from all tests until the end of the test run

@peter-ahe-google @wibling

peter-ahe-google, 2015-11-16:
My gut reaction is that we should prune the log when calling "show log". This could be using an option.

Also, we should call show log *after* running the test, not before.

Alternatively, we can limit how many commands can execute in one session.

Let's talk more in person.

mkustermann, 2015-11-16:
I've been debugging this a bit more - dumping memory directly from `/proc/<pid>/mem` and it looks, that it's not necessarily the 'fletch show log', most of the data is of the following form:
```
....
out/DebugXARM/fletch attach tcp_socket 127.0.0.1:54721 in session 0
out/DebugXARM/fletch export /home/chrome-bot/kustermann/fletch/out/DebugXARM/generated_tests/language/method_override3_test_01.dart to file /home/chrome-bot/kustermann/fletch/out/DebugXARM/generated_compilations/fletchc/multitest_language_method_override3_test_01/fletch.snapshot in session 0
out/DebugXARM/fletch attach tcp_socket 127.0.0.1:49377 in session 0
out/DebugXARM/fletch export /home/chrome-bot/kustermann/fletch/out/DebugXARM/generated_tests/language/prefix_assignment_test_02.dart to file /home/chrome-bot/kustermann/fletch/out/DebugXARM/generated_compilations/fletchc/multitest_language_prefix_assignment_test_02/fletch.snapshot in session 0
...
```

Another observation:
Using the observatory I sometimes see that there are 1024 byte Uint8List's which contained the string "Fletch VM exitcode is [number]". This message is coming from `tools/testing/dart/fletch_session_command.dart`:
```
194         fletch.stderr.writeln("Fletch VM exitcode is $vmExitCode");
```
I'm unsure why we end up using **1 KB** buffers for such a small strings. It might be the default buffer size for a utf8 encoder?

sgjesse, 2015-11-17:
Not sure if you already checked, but yes the default buffer for the UTF8 encode is 1K (https://github.com/dart-lang/sdk/blob/master/sdk/lib/convert/utf.dart#L141). However I would assume that this will be very short-lived buffers.

@lrhn

lrhn, 2015-11-17:
Yes, that's likely a `_Utf8EncoderSink`. 

I don't know why converting a single string uses the chunked conversion sink. Maybe it's a sink used by stderr, which could keep it alive.

mkustermann, 2015-11-18:
We've identified now the root cause of the memory problems (ignoring the other smaller things, like utf8 encoder buffer size): tools/test.py  has quadratic memory behavior in the number of tests we're running.

One solution would be: https://codereview.chromium.org/1459573002

closed 2015-11-19 by mkustermann