dart-archive/sdk #325: "fletch run hello.dart" tries to reuse fletch-vm even if it's already dead

mkustermann, 2015-11-20:
Assuming "foo.dart" is a program that will just keep running until we <ctrl>-C it, the following happens

```
$ fletch run foo.dart
... output from foo.dart ...
<ctrl>-C
$ fletch run foo.dart
Bad state: Trying to send command CommandCode.NewMap(MapId.methods) to fletch-vm, but the connection is already closed.
#0      FletchVmSession.sendCommand.<sendCommand_async_body> (package:fletchc/session.dart:133:7)
#1      Future.Future.microtask.<anonymous closure> (dart:async/future.dart:144)
#2      _rootRun (dart:async/zone.dart:904)
#3      _CustomZone.run (dart:async/zone.dart:803)
#4      _CustomZone.bindCallback.<anonymous closure> (dart:async/zone.dart:736)
#5      _microtaskLoop (dart:async/schedule_microtask.dart:43)
#6      _microtaskLoopEntry (dart:async/schedule_microtask.dart:52)
#7      _runPendingImmediateCallback (dart:isolate-patch/isolate_patch.dart:96)
#8      _RawReceivePortImpl._handleMessage (dart:isolate-patch/isolate_patch.dart:149)
```

@wibling @peter-ahe-google 

Labeled as Area-Compiler for now.

peter-ahe-google, 2015-11-20:
This is probably a side-effect of not null'ing out the debugger session to support live programming, etc.

mkustermann, 2015-11-20:
Thanks for taking a look.

Generally speaking, it would be nice if we could have such "smoke tests" for the "fletch" command line interface. Maybe it's worth filing another issue for that?

peter-ahe-google, 2015-11-20:
I usually add regression tests when I fix an issue. `tools/testing/dart/fletch_session_command.dart` already serves as a smoke test, but sometimes people make mistakes.

wibling, 2015-11-20:
Good idea. I will file an issue for that.

@ahe would this be a sideeffect of change
https://codereview.chromium.org/1439313003/?

wibling, 2015-11-20:
I have filed https://github.com/dart-lang/fletch/issues/328.

peter-ahe-google, 2015-11-20:
@wibling. Yes, I think the root cause is CL 1439313003. I think we explicitly need to `state.session = null` when the session encounters an error.

wibling, 2015-11-20:
Okay. Thanks for the info.

closed 2015-11-23 by wibling