dart-archive/sdk #328: Create fletch cmdline tool smoke tests

wibling, 2015-11-20:
We should have fletch cmdline tool smoke tests that test rudimentary commands for the tool.

E.g.
 fletch run foo.dart
 fletch show log
 ...

Also add running the same command a couple of times. See https://github.com/dart-lang/fletch/issues/325 for an example.

zerny, 2015-11-24:
@peter-ahe-google 
These tests should be part of our fletch_tests setup, but should we run tests by mocking the process and directly using Action objects or should we start a subprocess and interact using the command line interface?

peter-ahe-google, 2015-11-30:
@zerny that depends on what you want to test. I think you'd want some tests that are based on running the command line (black box), other tests that are using Action objects, and other tests that are calling methods in developer.dart so they can ensure that SessionState.current has the correct state before and after the operation (white box).

wibling, 2015-12-23:
This seems to be a general placeholder for adding more fletch command line tests.

Assigning it to Ian as he has created the CLITest et. al.