dart-archive/sdk #332: crash when running from RO memory (aka flash)

cpu-chromium, 2015-11-24:
Sorry for the complicated bug... if you get bored or dizzy skip all the way to #@#

For the STM32F7 disco board we can run unfolded snapshots from qspi-flash (serial flash) when we put said memory in "linear mode", in this mode an internal controller emulates memory access and under the covers does a block read. 

Here is the hacky code in SoD that does that
https://codereview.chromium.org/1441953002/

That works nicely but the controller also 'fakes' writes, that is the writes end up in memory but not ever written to flash, in general the qspi in linear mode is not writable, so if one applies this patch to set that memory region read-only:

diff --git a/platform/stm32f7xx/init.c b/platform/stm32f7xx/init.c
index 9e05cf6..2f218ef 100644
--- a/platform/stm32f7xx/init.c
+++ b/platform/stm32f7xx/init.c
@@ -256,6 +256,23 @@ static void mpu_init(void)
     HAL_MPU_ConfigRegion(&MPU_InitStruct);
 #endif
 
+    // QSPI (in linear mode)
+#if 1
+    MPU_InitStruct.Enable = MPU_REGION_ENABLE;
+    MPU_InitStruct.BaseAddress = QSPI_BASE;
+    MPU_InitStruct.Size = MPU_REGION_SIZE_16MB;
+    MPU_InitStruct.AccessPermission = MPU_REGION_PRIV_RO;
+    MPU_InitStruct.IsBufferable = MPU_ACCESS_BUFFERABLE;
+    MPU_InitStruct.IsCacheable = MPU_ACCESS_CACHEABLE;
+    MPU_InitStruct.IsShareable = MPU_ACCESS_NOT_SHAREABLE;
+    MPU_InitStruct.Number = region_num++;
+    MPU_InitStruct.TypeExtField = MPU_TEX_LEVEL1;
+    MPU_InitStruct.SubRegionDisable = 0x00;
+    MPU_InitStruct.DisableExec = MPU_INSTRUCTION_ACCESS_ENABLE;
+
+    HAL_MPU_ConfigRegion(&MPU_InitStruct);
+#endif
+
     HAL_MPU_Enable(MPU_HFNMI_PRIVDEF);
 }

Then fletch crashes, trying to do a memset, after the fletch program ran, during teardown:

Approximately line 430 of
https://github.com/dart-lang/fletch/blob/e9e449e18103f58afab0c51f99c4b12a32cb3a1d/src/vm/object_memory.cc

#@#
What I think would be practical is to construct a linux test that runs an unfolded snapshot in a read-only mmaped set of pages. It should cause the same crash.

madsager, 2015-11-24:
Yeah, that makes sense. We zap all deallocated pages in debug mode to make it easier to catch bugs; I'm guessing this is what you are seeing here. If you build with 'DEBUG= make' to get the release version, does it still crash?

In any case we should avoid doing that when running from read-only memory. I'll have a look. Thanks!

madsager, 2015-11-24:
We definitely need to add some testing of relocated program heaps. It seems that we are setting up a standard Fletch Chunk for the memory as well. That also means that we end up calling free/page_free on the memory during shutdown which we shouldn't do either.

@herhut-ggl

closed 2015-11-24 by madsager