dart-archive/sdk #335: Test framework reporting crash (255) in spite of vm reporting Compile error

sigurdm, 2015-11-25:
To reproduce make a test (eg. `tests/unsorted/foo_test.dart`)
```
import "package:async_helper/async_helper.dart";

test() {
  ***; // Compile time error.
}

main() {
  asyncStart();
  test();
}
```

Run it with the test.py tool:
```
$ tools/test.py unsorted/foo
Running: gclient runhooks
Use --run-gclient-hooks=0 to skip this step.

________ running 'download_from_google_storage --no_auth --no_resume --bucket dart-dependencies-fletch -d -r --auto_platform fletch/third_party/bin' in '/usr/local/google/home/sigurdm/projects/fletch-repo'
0> File fletch/third_party/bin/linux/dart-arm exists and SHA1 matches. Skipping.
0> File fletch/third_party/bin/linux/dart exists and SHA1 matches. Skipping.
Success!
Downloading 2 files took 0.046414 second(s)

________ running 'download_from_google_storage --no_auth --no_resume --bucket dart-dependencies-fletch -d -r --auto_platform fletch/tools/testing/bin' in '/usr/local/google/home/sigurdm/projects/fletch-repo'
0> File fletch/tools/testing/bin/linux/dart-arm exists and SHA1 matches. Skipping.
0> File fletch/tools/testing/bin/linux/dart exists and SHA1 matches. Skipping.
Success!
Downloading 2 files took 0.047042 second(s)

________ running 'download_from_google_storage --no_auth --no_resume --bucket dart-dependencies-fletch -d fletch/pkg/mdns/lib/native' in '/usr/local/google/home/sigurdm/projects/fletch-repo'
0> File fletch/pkg/mdns/lib/native/libmdns_extension_lib.so exists and SHA1 matches. Skipping.
0> File fletch/pkg/mdns/lib/native/libmdns_extension_lib.dylib exists and SHA1 matches. Skipping.
Success!
Downloading 2 files took 0.001528 second(s)

________ running '/usr/bin/python fletch/tools/vs_dependency/vs_toolchain.py update' in '/usr/local/google/home/sigurdm/projects/fletch-repo'

________ running 'download_from_google_storage --no_auth --no_resume --bucket dart-dependencies-fletch -d -r -u --auto_platform fletch/third_party/qemu' in '/usr/local/google/home/sigurdm/projects/fletch-repo'
0> File fletch/third_party/qemu/linux/qemu.tar.gz exists and SHA1 matches. Skipping.
Success!
Downloading 1 files took 0.006058 second(s)

________ running 'download_from_google_storage --no_auth --no_resume --bucket dart-dependencies-fletch -d -r -u --auto_platform fletch/third_party/openocd' in '/usr/local/google/home/sigurdm/projects/fletch-repo'
0> File fletch/third_party/openocd/linux/openocd.tar.gz exists and SHA1 matches. Skipping.
Success!
Downloading 1 files took 0.010985 second(s)

________ running '/usr/bin/python fletch/tools/clang_update.py --revision=245965' in '/usr/local/google/home/sigurdm/projects/fletch-repo'
[linux] Updating Clang to 245965-1 ...
Already up to date.
[mac] Updating Clang to 245965-1 ...
Already up to date.

________ running 'ninja -C fletch' in '/usr/local/google/home/sigurdm/projects/fletch-repo'
ninja: Entering directory `fletch'
ninja: no work to do.
Test configurations: none_fletch_warnings_debug_ia32 none_fletch_tests_debug_ia32 none_fletchc_debug_ia32 none_fletch_cc_tests_debug_ia32
Building in out/DebugIA32.
Use --build-before-testing=0 to skip this step.
Building done.
Total: 1 tests
 * 0 tests will be skipped (0 skipped by design)
 * 0 tests are expected to be flaky but not crash
 * 1 tests are expected to pass
 * 0 tests are expected to fail that we won't fix
 * 0 tests are expected to fail that we should fix
 * 0 tests are expected to crash that we should fix
[... 164 lines cut ...]
     is the easiest way to debug using both gdb and lldb. You need to start two
     processes, each in their own terminal window:

    gdb -ex run --args out/DebugIA32/fletch-vm --port=54321

    out/DebugIA32/dart -Dtest.fletch_settings_file_name=.fletch-settings \
      -Dfletch.version=`out/DebugIA32/fletch-vm --version` \
      -c -DattachToVm=54321 tests/fletchc/run.dart /usr/local/google/home/sigurdm/projects/fletch-repo/fletch/tests/unsorted/foo_test.dart

Took 0:00:04.370880

Short reproduction command (experimental):
    python tools/test.py -rfletchc -t120 unsorted/foo_test

===
=== 1 test failed
===

[00:04 | 100% | +    0 | -    1]

--- Total time: 00:04 ---
0:00:04.370880 - fletch_session - none-fletchc debug_ia32/unsorted/foo_test
```

The problem does not appear if I run `tools/test.py unsorted/foo -rfletchvm -cfletchc`.

mkustermann, 2015-11-25:
I think this will get fixed with another CL that is waiting to be reviewed: https://codereview.chromium.org/1453913004/

The problem is that `asyncStart()` starts another process which might die (with uncaught signal, which is like uncaught exception, which is 255) after main() died.

peter-ahe-google, 2015-11-30:
It's strange that this is reported as a crash. That sounds like a bug in test framework and not related to [CL 1453913004](https://codereview.chromium.org/1453913004/).

mkustermann, 2015-11-30:
To be more precise, what I ment is that even though `fletch run` returned **254** the fletch-vm exit code was **255**. This problem should be solved with my CL (which already landed).

The fletch-vm exit code depends on the ordering of death of the processes (aka isolates). Since the fletch `asyncStart` implementation uses a new isolate - which could die later than the main isolate with the compile-time error, it will be reported as an unhandled signal (255).

My change will now ensure that processes (aka isolates) are viewed as a tree and the exit code will correspond to the result of the main isolate (even if it dies earlier than others).

=> So we get now **254** in both cases. So the original issue has been fixed.

(Though I'm not sure, why the testing framework would consider 255 as a crash)

mit-mit, 2015-12-07:
@mkustermann can we close this bug?

mkustermann, 2015-12-10:
The original issue has been fixed (the test outcome will be CompileTimeError), so I think we should.

closed 2015-12-10 by mkustermann