dart-archive/sdk #338: Enable the user to see warnings in a package they are developing

mit-mit, 2015-11-27:
I'm developing a new package. This includes a small sample that uses the package. Often when I run the sample program, Fletch me that there are suppresed warnings in the package:

```
$ ./out/ReleaseX64/fletch run pkg/mqtt/subscribe-sample.dart 
Hint: 12 warning(s) suppressed in package:mqtt.
```

It would be great to have a way of seeing those.

kasperl, 2015-11-27:
For now, I think I'd be happy with always showing all warnings -- and we can achieve that by adding '--show-package-warnings' to the `EXTRA_DART2JS_OPTIONS` list:

https://github.com/dart-lang/fletch/blob/master/pkg/fletchc/lib/src/fletch_compiler_implementation.dart#L65

mit-mit, 2015-11-27:
Yes, that works great. Iff we want a flag on the fletch, I think the default should be to show these and then have a suppress flag.

commit 6fadd7b9 (2015-11-27): Show package warnings.

Closes #338

R=mit@google.com

Review URL: https://codereview.chromium.org/1477333002 .
closed 2015-11-27 by karlklose