dart-archive/sdk #340: Sometimes relative paths in GYP ldflags section is "off by one"

sgjesse, 2015-11-30:
In `platforms/stm/disco_fletch.gyp` we have the following comment:

```
'ldflags': [
  '-specs=nosys.specs',
  '-specs=nano.specs',
  # TODO(sgjesse): Why does this not work???
  #'-T<(generated_path)/SW4STM32/configuration/STM32F746NGHx_FLASH.ld',
  # TODO(sgjesse): Why is this needed???
  '-T../../platforms/stm/disco_fletch/generated/SW4STM32/'
    'configuration/STM32F746NGHx_FLASH.ld'
],
```

The variable `generated_path` _is_ `../../platforms/stm/disco_fletch/generated/`, and works in other rules, but in `ldflags` an additional `..` is added to the generated project.

peter-ahe-google, 2015-11-30:
You have to put a relative file name in a variable named something ending with "_dir" or "_file". See [Pathname Relativization](https://gyp.gsrc.io/docs/InputFormatReference.md#Pathname-Relativization).

sgjesse, 2015-12-11:
I cannot get pathname relativization to work for `ldflags`.

Asked on gyp-developer: https://groups.google.com/forum/#!topic/gyp-developer/W_yZcIsFXEc.