dart-archive/sdk #349: We need to make new/new[] never return NULL pointer

mkustermann, 2015-12-04:
On LK we new/new[] might return NULL pointers and since we always assume it works, we sometimes run OOM and go into an infinite loop. This causes some of our LK tests to time out. One test was marked as flaky in f6fb59079b2966105725aedd4dd28653f0fdfe0c.

This could be fixed in different ways. For example by 
* making sure LK itself never returns NULL from new/new[]
* marking LK not go into an infinite loop on 0-deref but rather abort, thereby making tests crash and not timeout
* having our own new/new[] and abort if malloc failed

cc: @ErikCorryGoogle

cpu-chromium, 2015-12-04:
LK tries to handle OOM gracefully. Can you point to the particular cases you are hitting?

travisg, 2015-12-04:
I'll check into the infinite loop. For qemu based runs there's probably
some sort of way to tell the emulator to abort and shutdown, and it might
make sense to (optionally) set some build flag that says as much so tests
like this can catch it.

mkustermann, 2015-12-11:
Sorry for getting back to you so late.

Here is what we've been observing:

When running our tests with lk, our testing infrastructure will make cross-platform-compatible snapshots and run them using qemu and a lk fletch build. Sometimes 

>  $ tools/lk/run_snapshot_lk_qemu.sh  out/DebugIA32/generated_compilations/fletchc/tests_unsorted_messaging_process_die_early_gc_on_>  delete_test/fletch.snapshot        Starting qemu...
>  Started with PID 4240
>  Waiting for qemu to come up...
>  QEMU 2.4.1 monitor - type 'help' for more information
>  (qemu) Starting fletch...
>  Waiting for size...
>  Sending size (33482)...
>  Waiting for snapshot request...
>  Sending snapshot...
>  ................................
>  SNAPSHOT READ.
>  STARTING fletch-vm...
>  LOADING snapshot...
>  RUNNING program...
>  
>  cpu 0 data abort, translation fault on write
>  DFAR 0x0 (fault address)
>  DFSR 0x805 (fault status register)
>  halting
>  current_thread 0x80098a50, name Dart thread
>  **r0  0x00000000** r1  0x800a09e8 r2  0x80abd618 r3  0x00000001
>  r4  0x80abd618 r5  0x00000000 r6  0x00000000 r7  0x800a09e8
>  r8  0x00002842 r9  0x800a09e8 r10 0x80abd618 r11 0x00000000
>  r12 0x00000ff1 usp 0x00000000 ulr 0x00000000 **pc  0x800343a6**
>  spsr 0x60000173
>   usr r13 0x00000000 r14 0x00000000
>   fiq r13 0x00000000 r14 0x00000000
>   irq r13 0x00000000 r14 0x8005d63c
>  asvc r13 0x80099a50 r14 0x8001104f
>  *svc r13 0x80099b28 r14 0x800327df
>   und r13 0x00000000 r14 0x00000000
>   sys r13 0x00000000 r14 0x00000000
>  bottom of stack at 0x80099b28:
>  0x80099b28: 80105670 80081168 8010568d 000000b8 |pV..h....V......|
>  0x80099b38: 800327d5 00002842 800a09e8 80abd618 |.'..B(..........|
>  0x80099b48: 00000000 800a09e8 80abd618 00002842 |............B(..|
>  0x80099b58: 800a09e8 00000001 00000001 800327df |.............'..|
>  ...
>  CRASH: starting debug shell... (reason = 9)
>  **entering panic shell loop**

This seems to be a NULL pointer dereference in a constructor:

>  80034388 <_ZN6fletch7ProcessC1EPNS_7ProgramEPS0_>:
>  80034388: b089        sub sp, #36 ; 0x24
>  8003438a: e9cd be07   strd  fp, lr, [sp, #28]
>  8003438e: e9cd 9a05   strd  r9, sl, [sp, #20]
>  80034392: e9cd 7803   strd  r7, r8, [sp, #12]
>  80034396: e9cd 5601   strd  r5, r6, [sp, #4]
>  8003439a: 9400        str r4, [sp, #0]
>  8003439c: b087        sub sp, #28
>  8003439e: 4683        mov fp, r0
>  800343a0: 460f        mov r7, r1
>  800343a2: 4692        mov sl, r2
>  800343a4: 2600        movs  r6, #0
>  **800343a6: 6006        str r6, [r0, #0]**
>  800343a8: 6046        str r6, [r0, #4]
>  800343aa: f8cb 1008   str.w r1, [fp, #8]
>  800343ae: 60c6        str r6, [r0, #12]
>  800343b0: 680b        ldr r3, [r1, #0]
>  800343b2: 6103        str r3, [r0, #16]
>  800343b4: 6146        str r6, [r0, #20]

So the call is inside a C++ object constructor (namely `fletch::Process::Process(fletch::Program*, fletch::Process*)`) - but this is different in different times. This made us believe that

* the `new` operator sometimes simply returns `NULL`, 
* we run into a NULL-deref
* the crash shell loops forever
* our test will timeout

Our codebase assumes that the `new` operator does always work and never returns `NULL`. Therefore I think we should put in a `FATAL("Unable to allocate memory");` or something instead of letting it just crash at random places when reading/writing to the memory.

It's also important that our tests will just crash instead of hang in a loop and time out. This makes our test cycle times longer.

Our normal implementation of `FATAL("...")` will make sure our tests will be considered crashing and will not timeout.

@cpu-chromium @travisg Any ideas on how we should approach this?