dart-archive/sdk #353: VM: Only store methods for instantiated classes in dispatch table

kasperl, 2015-12-07:
Today, the dispatch table is a compressed form of the M x N table where M is the number of classes and N is the number of selectors.

We should be able to make it a compressed form of the M' x N table where M' is the number of instantiated classes. This way, the dispatch table will only contain entries that are useful because the class of a receiver in an ordinary method invoke can only be an instantiated class -- not one of its super classes.

Concretely, I suggest numbering the instantiated classes from 0 to M'-1 and let the remaining ids for the remaining classes start at M'. We can find the instantiated classes by looking for 'allocate' bytecodes in the program space.

kasperl, 2015-12-07:
@skabet might be interested in taking a look at this.

skabet, 2015-12-08:
I have a working version locally, will clean up tomorrow.