dart-archive/sdk #355: Running a program that throws leaves the VM in an illegal state.

herhut-ggl, 2015-12-07:
When running 

```
main () {
  throw 42;
}
```

on the first round, the VM does the right thing:

```
Uncaught exception: 42
Stack trace:
> 0: main        @42

usr/local/google/home/herhut/src/dart/bug153/nsm.dart:2:9
1    main () {
2      throw 42;
3    }
```

However, invoking the run command again without first performing a quit leads to 

```
../../src/vm/session.cc:1032: error: expected: !program()->is_compact()
Bad state: Trying to send command CommandCode.PushFromMap(MapId.methods, 9) to fletch-vm, but the connection is already closed.
#0      FletchVmSession.sendCommand.<sendCommand_async_body> (package:fletchc/session.dart:138:7)
#1      Future.Future.microtask.<anonymous closure> (dart:async/future.dart:144)
#2      _rootRun (dart:async/zone.dart:907)
#3      _CustomZone.run (dart:async/zone.dart:802)
#4      _CustomZone.bindCallback.<anonymous closure> (dart:async/zone.dart:735)
#5      _microtaskLoop (dart:async/schedule_microtask.dart:43)
#6      _microtaskLoopEntry (dart:async/schedule_microtask.dart:52)
#7      _runPendingImmediateCallback (dart:isolate-patch/isolate_patch.dart:96)
#8      _RawReceivePortImpl._handleMessage (dart:isolate-patch/isolate_patch.dart:149)
```

ErikCorryGoogle, 2015-12-09:
I've spent too much time on this and I'm not making progress.
Someone else is welcome to pick up https://codereview.chromium.org/1506763005/

wibling, 2015-12-10:
This should be fixed in 0b1ecc675c62fc4e00997e2b4bd7250bffbd26a0

wibling, 2015-12-23:
Now with test and a fix for the interactive debugger as well, 33fc0325eadc35037cb9f90979751a1c0657f8b4.

closed 2015-12-23 by wibling