dart-archive/sdk #366: Persistent process ignores changes to packages if not restarted

ricowind, 2015-12-15:
@peter-ahe-google @wibling @sgjesse 

Version: 0.2.0-edge.50ffce48a0d44ac290c2623543a01d5051039c71
OS: Linux

How to reproduce the problem:
Create a test.dart file with the following content:
```
import 'package:file/file.dart';

main() {
  var f = new File.open("/tmp/foobar.txt", mode: File.WRITE);
  f.close();
}
```

As expected, the output of running:
```
$ out/DebugX64/fletch run test.dart
```
is no output on the console

Add this line in pkg/file/lib/file.dart in the close method:
```
    print("closing file");
```
Run:
```
$ out/DebugX64/fletch run test.dart
```
Still no output.

Run:
```
$ out/DebugX64/fletch quit
$ out/DebugX64/fletch run test.dart
closing file
```
we now have output.

sgjesse, 2015-12-15:
Can this be related to incremental compilation? Can you try to change

```
"incremental_mode": "production"
```

to

```
"incremental_mode": "none"
```

in the `.fletch-settings` file?

ricowind, 2015-12-15:
@sgjesse same result with incremental_mode set to none