dart-archive/sdk #373: Fletch build image support

mit-mit, 2015-12-15:
Build flashable image from fletch program

ricowind, 2016-02-10:
Closing this, sgjesse made this possible with the dartino x-build command

mit-mit, 2016-02-11:
Re-opening, we should not close this until x-build has been promoted to build

ricowind, 2016-02-11:
Promoted to build?
This is already in the latest dev channel build, but it has not been promoted to users yet if that is what you mean? (if that is what you mean, do you expect us to not close all the bugs we fix until it has shipped to users? - that would suck, since you have no progress indication)

mit-mit, 2016-02-11:
By promoted, I mean renamed from x-build to build. I thought x- was used
for things that were not completed yet.

sgjesse, 2016-02-11:
The `x-` prefix is for commands which are still under development/in beta. I don't think we will remove the `x-` prefixes from these commands anytime soon - unless we decide either

- these commands are probably going to stick
- or we don't care about changing the commands between releases

@lukechurch What is your take on this?

mit-mit, 2016-02-12:
@sgjesse I discussed with @lukechurch, and we agree that this naming would be great:

```
dartino build <file name>: build flashable image containing the Dart code in <file name>

dartino flash <file name>: build flashable image containing the Dart code in <file name>,
   and flash it to a connected STM board
```

Longer term we probably need to add support for flashing pre-built images, and for specifying which device to flash to (e.g., when multiple supported ones are connected), but that can follow later.

commit 914d40ae (2016-02-15): Rename x-build to build and x-deploy to flash

Closes #373

BUG= https://github.com/dartino/sdk/issues/373
R=ricow@google.com

Review URL: https://codereview.chromium.org/1701663002 .
closed 2016-02-15 by sgjesse