dart-archive/sdk #375: Fletch crashes with session remote: worker isolate terminated unexpectedly

filiph, 2015-12-16:
After some use, running anything (even samles/general/hello.dart) in session remote immediately throws the internal error below.

OS: Mac OSX Yosemite 10.10.5 (14F1021)
Fletch SDK version (0.2.0-dev.4.5)
Using the recommended setup, with SenseHat.

Internal error: session remote: worker isolate terminated unexpectedly
#0      throwInternalError (package:fletchc/src/diagnostic.dart:187:3)
#1      DriverVerbContext.performTaskInWorker.<performTaskInWorker_async_body> (package:fletchc/src/driver/driver_main.dart:318:7)
#2      Future.Future.microtask.<anonymous closure> (dart:async/future.dart:144)
#3      _rootRun (dart:async/zone.dart:900)
#4      _CustomZone.run (dart:async/zone.dart:803)
#5      _CustomZone.runGuarded (dart:async/zone.dart:709)
#6      _CustomZone.bindCallback.<anonymous closure> (dart:async/zone.dart:734)
#7      _rootRun (dart:async/zone.dart:904)
#8      _CustomZone.run (dart:async/zone.dart:803)
#9      _CustomZone.runGuarded (dart:async/zone.dart:709)
#10     _CustomZone.bindCallback.<anonymous closure> (dart:async/zone.dart:734)
#11     _microtaskLoop (dart:async/schedule_microtask.dart:43)
#12     _microtaskLoopEntry (dart:async/schedule_microtask.dart:52)
#13     _Timer._runTimers (dart:isolate-patch/timer_impl.dart:394)
#14     _Timer._handleMessage (dart:isolate-patch/timer_impl.dart:414)
#15     _RawReceivePortImpl._handleMessage (dart:isolate-patch/isolate_patch.dart:148)

Fletch can find the device with `fletch show devices`.

filiph, 2015-12-16:
Restart of device doesn't help.

filiph, 2015-12-16:
Ok, running `fletch quit` fixed this.

ricowind, 2015-12-17:
There has been some work on this lately, by I am not sure if this specific problem is fixed @wibling