dart-archive/sdk #376: Running the persistent compiler twice on the same code will give good warnings only the first time

ricowind, 2015-12-17:
@peter-ahe-google 

This may actually be two separate issues, but I am not sure.

OS: linux
Version: 0.2.0-edge.50ffce48a0d44ac290c2623543a01d5051039c71

How to reproduce the problem:
Create a dart file test.dart:
```
int foobar(int a, int b) {
  return a + b;
}

void main() {
  print(foobar(42));
}
```

Run fletch export on it:
$ ~/src/fletch/fletch/out/DebugX64/fletch export test.dart to file snapshot
usr/local/google/home/ricow/src/mbedtls/mbedtls/test.dart:8:9:
Warning: Missing argument of type 'int'.
  print(foobar(42));
        ^^^^^^^^^^
usr/local/google/home/ricow/src/mbedtls/mbedtls/test.dart:3:1:
Info: This is the method declaration.
int foobar(int a, int b) {
^^^^^^^^^^^^^^^^^^^^^^^^
```
All good. Now run the compiler again (or do a whitespace change in the test.dart file):
```
$ ~/src/fletch/fletch/out/DebugX64/fletch export test.dart to file snapshot

```
No output, should it not output the same warnings?

Now actually do something that changes the code, e.g., add a 'int c' parameter to the foobar function, export again:
```
~/src/fletch/fletch/out/DebugX64/fletch export test.dart to file snapshot
file:///usr/local/google/home/ricow/src/mbedtls/mbedtls/test.dart+78-88: Missing argument of type 'int'.
file:///usr/local/google/home/ricow/src/mbedtls/mbedtls/test.dart+2-33: This is the method declaration.
```
The warnings now get shown, but in a way less informative way (i.e., as I user I have no idea what those numbers mean)

ricowind, 2015-12-17:
also @sigurdm @karlklose may know something about this

sigurdm, 2015-12-17:
It seems like the input-provider is wrong, or its path is wrong in the last run. That is probably my mistake.

peter-ahe-google, 2015-12-17:
This is a known problem in incremental compilation. It's the area I'm currently focusing on, and there are two separate issues here:

* Tracking correct positions across incremental compilations

* Managing diagnostics across program changes. For example, there's an error in method `a` because it tries to call method `b` that doesn't exist. Now you add method `b`. In this case, you want the error in `a` to go away, but `a` wasn't modified. On the other hand, if you didn't add `b`, but `c`, then you want to repeat the error in `a` when the program is recompiled.

ricowind, 2015-12-17:
I have  "incremental_mode": "none" in my .fletch-settings file in my checkout, but I still see this error

peter-ahe-google, 2015-12-17:
I'm not entirely sure how `.fletch-settings` is read now. @wibling may have changed it. However:

* .fletch-settings is read from your current working directory (unless Gustav changed this)

* The special sessions `local` and `remote` do not read `.fletch-settings` at all. They use a built-in default.

* If you don't specify a session name, the default used is `local` (which ignores .fletch-settings).

Try running this command:

    fletch show log

If it prints something like this:

```
[local: 2015-12-17 12:57:05.054 created session with settings Settings(packages: file:///Users/ahe/Dart/fletch-repo/fletch/.packages, options: [], constants: {}, device_address: null, device_type: DeviceType.mobile, incremental_mode: IncrementalMode.none)]
```

Then incremental mode is indeed turned off.

However, the messages we're seeing may still be due to a bug in the incremental compiler's diagnostic listener. It will definitely produce output like this, but there can be other reasons.

ricowind, 2015-12-17:
ok, so that prints:
```
 ~/src/fletch/fletch/out/DebugX64/fletch show log
[local: 2015-12-17 12:49:40.712 created session with settings Settings(packages: file:///usr/local/google/home/ricow/src/fletch/fletch/.packages, options: [], constants: {}, device_address: null, device_type: DeviceType.mobile, incremental_mode: IncrementalMode.production)]
[local: 2015-12-17 12:49:43.182 Compiled 'file:///usr/local/google/home/ricow/src/mbedtls/mbedtls/test3.dart' to 3251 commands]
[local: 2015-12-17 12:49:43.222 Connected to TCP vmSocket  vmSocket 45333 -> 127.0.0.1:40163]
[local: 2015-12-17 12:49:50.519 Compiling difference from file:///usr/local/google/home/ricow/src/mbedtls/mbedtls/test3.dart to file:///usr/local/google/home/ricow/src/mbedtls/mbedtls/test3.dart]
[local: 2015-12-17 12:49:50.532 Reusing origin library(dart:core) (assumed read-only).]
[local: 2015-12-17 12:49:50.532 Reusing library(dart:fletch._system) (assumed read-only).]
[local: 2015-12-17 12:49:50.532 Reusing origin library(dart:_internal) (assumed read-only).]
[local: 2015-12-17 12:49:50.533 Reusing origin library(dart:collection) (assumed read-only).]
[local: 2015-12-17 12:49:50.533 Reusing origin library(dart:math) (assumed read-only).]
[local: 2015-12-17 12:49:50.533 Reusing library(dart:fletch) (assumed read-only).]
[local: 2015-12-17 12:49:50.533 Reusing origin library(dart:convert) (assumed read-only).]
[local: 2015-12-17 12:49:50.533 Reusing origin library(dart:async) (assumed read-only).]
[local: 2015-12-17 12:49:50.533 Reusing library(dart:fletch.os) (assumed read-only).]
[local: 2015-12-17 12:49:50.533 Reusing library(dart:fletch.ffi) (assumed read-only).]
[local: 2015-12-17 12:49:50.533 Reusing origin library(dart:typed_data) (assumed read-only).]
[local: 2015-12-17 12:49:50.537 Attempting to reuse library(file:///usr/local/google/home/ricow/src/mbedtls/mbedtls/test3.dart).]
[local: 2015-12-17 12:49:50.537 New library synthesized.]
[local: 2015-12-17 12:49:50.540 Differences computed.]
[local: 2015-12-17 12:49:50.540 Looking at difference: Modified(function(foobar) -> function(foobar))]
[local: 2015-12-17 12:49:50.542 Removed method function(foobar).]
[local: 2015-12-17 12:49:50.545 Signature change requires 'experimental' mode]
[local: 2015-12-17 12:49:50.550 Can't incrementally compile program.

In function(foobar):
  Signature change requires 'experimental' mode]
[local: 2015-12-17 12:49:50.550 Attempting full compile...]
[local: 2015-12-17 12:49:51.897 Compiled 'file:///usr/local/google/home/ricow/src/mbedtls/mbedtls/test3.dart' to 3251 commands]
[local: 2015-12-17 12:49:51.902 Connected to TCP vmSocket  vmSocket 41156 -> 127.0.0.1:57804]
```
How do I disable incremental compilation then for my default local session?

sgjesse, 2015-12-17:
Change ~/local.fletch-settings

peter-ahe-google, 2015-12-17:
What @sgjesse said. Remember to restart the persistent process or end the session:

* Restart persistent process: `fletch quit`

* End the session: `fletch x-end session local`

Settings files aren't currently automatically reloaded.