dart-archive/sdk #379: Simple unresolved getter will crash the compiler

ricowind, 2015-12-17:
@sigurdm @peter-ahe-google 
OS: linux
Version: 0.2.0-edge.e21339a0f7b7ae9d545cb9cdb50f4b93873b432b

How to reproduce the problem:
Create a test file with:
```
void main() {
  foobar;
}
```
Expected result: warning or compile time error that foobar can't be resolved

Actual output:
```~/src/fletch/fletch/out/DebugX64/fletch run test3.dart
test3.dart:2:3:
Warning: Cannot resolve 'foobar'.
  foobar;
  ^^^^^^
test3.dart:1:1:
Internal Error: The compiler crashed when compiling this element.
void main() {
^^^^^^^^^^^
The compiler is broken.

When compiling the above element, the compiler crashed. It is not
possible to tell if this is caused by a problem in your program or
not. Regardless, the compiler should not crash.

The Dart team would greatly appreciate if you would take a moment to
report this problem at http://dartbug.com/new.

Please include the following information:

* the name and version of your operating system,

* the Dart SDK build number (build number could not be determined), and

* the entire message you see here (including the full stack trace
  below as well as the source location above).

The Fletch program is broken and has crashed.

The Fletch team would greatly appreciate if you would take a moment to
report this problem at https://github.com/dart-lang/fletch/issues/new

Please include the following information:

* the name and version of your operating system

* the Fletch SDK version (0.2.0-edge.e21339a0f7b7ae9d545cb9cdb50f4b93873b432b)

* the entire message you see here (including the full stack trace below)
Class '_CompactLinkedHashSet<Element>' has no instance method '[]'.

NoSuchMethodError: method not found: '[]'
Receiver: Instance of '_CompactLinkedHashSet<Element>'
Arguments: [Instance of 'PartialMethodElement']
#0      Object._noSuchMethod (dart:core-patch/object_patch.dart:42)
#1      Object.noSuchMethod (dart:core-patch/object_patch.dart:45)
#2      CodegenVisitor.checkCompileError (package:fletchc/src/codegen_visitor.dart:3055:55)
#3      FunctionCodegen.compile (package:fletchc/src/function_codegen.dart:47:9)
#4      FletchBackend.codegenFunction (package:fletchc/src/fletch_backend.dart:974:15)
#5      FletchBackend.compileElement.<anonymous closure> (package:fletchc/src/fletch_backend.dart:802:9)
#6      _CompilerDiagnosticReporter.withCurrentElement (package:compiler/src/compiler.dart:1710:15)
#7      FletchDiagnosticReporter.withCurrentElement (package:fletchc/src/fletch_compiler_implementation.dart:333:30)
#8      FletchBackend.compileElement (package:fletchc/src/fletch_backend.dart:791:23)
#9      FletchEnqueuer.processQueue (package:fletchc/src/fletch_enqueuer.dart:219:40)
#10     FletchEnqueuer.forEach (package:fletchc/src/fletch_enqueuer.dart:199:5)
#11     Compiler.emptyQueue (package:compiler/src/compiler.dart:1106:11)
#12     Compiler.processQueue (package:compiler/src/compiler.dart:1134:5)
#13     Compiler.compileLoadedLibraries (package:compiler/src/compiler.dart:1057:5)
#14     Compiler.runInternal.<anonymous closure> (package:compiler/src/compiler.dart:877:7)
#15     _rootRunUnary (dart:async/zone.dart:914)
#16     _CustomZone.runUnary (dart:async/zone.dart:810)
#17     _Future._propagateToListeners.handleValueCallback (dart:async/future_impl.dart:502)
#18     _Future._propagateToListeners (dart:async/future_impl.dart:585)
#19     _Future._completeWithValue (dart:async/future_impl.dart:376)
#20     _Future._asyncComplete.<anonymous closure> (dart:async/future_impl.dart:430)
#21     _rootRun (dart:async/zone.dart:907)
#22     _CustomZone.run (dart:async/zone.dart:802)
#23     _CustomZone.bindCallback.<anonymous closure> (dart:async/zone.dart:735)
#24     _microtaskLoop (dart:async/schedule_microtask.dart:43)
#25     _microtaskLoopEntry (dart:async/schedule_microtask.dart:52)
#26     _runPendingImmediateCallback (dart:isolate-patch/isolate_patch.dart:96)
#27     _RawReceivePortImpl._handleMessage (dart:isolate-patch/isolate_patch.dart:149)
```

peter-ahe-google, 2015-12-17:
This looks like a Dart SDK issue. This class is defined by the Dart VM in `runtime/lib/compact_hash.dart` (`fletch/third_party/dart/runtime/lib/compact_hash.dart`) and it doesn't seem to implement operator[] (which must be a bug).

ricowind, 2015-12-17:
I did:
```
rm -rf third_party/dart
gclient sync
rm -rf out
ninja
ninja -C out/DebugX64
```
and ran again, now the compiler works. I have no idea how I got into that state, closing this

closed 2015-12-17 by ricowind