dart-archive/sdk #381: SD Card Init/Network Config script

butlermatt, 2015-12-17:
The scripts that setup the image on the raspberry pi are awesome. Great work!

One request I have for the network side though, would be to optionally setup a wifi device instead of ethernet and potentially even disable ethernet when setting up the wifi. As it stands now I have to manually track down the eth0 configuration and disable it and setup the wifi settings from the Pi. The eth0 being left on causes the bootup wait for 120 seconds looking for the ethernet that won't ever appear.

ricowind, 2015-12-17:
@sgjesse

butlermatt, 2015-12-17:
A couple other minor nits. It seems that the fletch setup the keyboard, locale and timezone to UK. Which with a standard US keyboard was.. unexpected. Not sure if there's a way to force the raspi-config to run or something similar to encourage the config.

Also, I had to disable the 'ip=xxx.xxx.xx.xx' part of /boot/cmdline.txt before I could resolve my boot issues.  I had originally provided a static IP during configuration. I then later re-ran the network config script to specify dhcp but the static IP remained in that part of the configuration.

ricowind, 2015-12-17:
Looking at the raspberry forums that seems standard for raspbian (e.g. https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=80127
https://www.youtube.com/watch?v=L1F-TxTPyiM
)

sgjesse, 2015-12-17:
Support for configuring Wi-Fi is something we did think about for the `flash_sd_card` and `configure-network`, but just never got to do for the first two releases. Opened #383 with that feature request.

Regarding the keyboard, as Rico mentioned, we are just providing the default Raspbian setup. The intention is (of course) that you should not need to attach a screen and keyboard. However if you do, running the [`raspi-config` script](http://elinux.org/RPi_raspi-config) should give you all the configuration options. You can even run `raspi-config` in a `ssh` terminal to configure your RPi (configuring the keyboard layout requires a direct connection though).

The fact that the `configure-network` script does not remove `ip=xxx.xxx.xx.xx` from `/boot/cmdline.txt` when DHCP is configured is a bug. Opened #382 to address that.

Closing this with new bugs for the remaining issues.

closed 2015-12-17 by sgjesse