dart-archive/sdk #384: FFI issues during debug

cpu-chromium, 2015-12-18:
From cousin project sod.

I have a piece of code that works correctly when it is just run and triggers an exception under the dart debugger. Here is the main code:
https://github.com/domokit/sod/blob/master/dart/examples/buttons/buttons_sync.dart

And the library is here
https://github.com/domokit/sod/blob/master/dart/pkg/lk/lib/ports.dart

Once the sod shell starts you, press enter and do:

] fletch buttons_sync.snap
waiting for buttons_sync.snap via TFTP. mode: run
starting fletch-vm...
loading snapshot: 34489 bytes ...
running program...
looping ..
[0] button: [0, 0, 0, 0, 1, 0, 0, 0]

This means it is reading the fake buttons press of line 40 of
https://github.com/domokit/sod/blob/master/app/fletch/sensors.c

But on the debugger it generates this exception

Starting session. Type 'help' for a list of commands.

> r
Uncaught exception: Instance of 'ArgumentError' {
  ArgumentError._hasValue: false
  ArgumentError.invalidValue: null
  ArgumentError.name: null
  ArgumentError.message: null
}
Stack trace:
> 0: LKReadPort._open lazy initializer   @ForeignLibrary.main.lookup('port_open')
  1: LKReadPort                          @_open
  2: main                                @new LKReadPort("sys/io/sw")
> q

===========
Repro steps
===========

in the sod repo, usual pull and gclient sync

$> make qemu-run

In the LK console, connected to your linux terminal/console

] fletch debug

Then in another terminal:
$> make fletch-debug-qemu ARGS=dart/examples/buttons/buttons_sync.dart

Then 'r' as usual.

One more thing to mention. Initially I thought that lines 53 -> 65 
https://github.com/domokit/sod/blob/master/app/fletch/loader.cpp

were the issue. However, I have tried several variations, including not exiting the thread and just sleeping after the connect. In the original code (before today) the thread 'tears down' fletch and exits. Regardless I get the same problem. What is the right shape of this code? should the debug thread really exit?

cpu-chromium, 2015-12-18:
Note sod's DEPS file, we are a bit behind.

ricowind, 2015-12-18:
I see this error even if just doing:
```
third_party/fletch/out/ReleaseIA32/fletch run dart/examples/buttons/buttons_sync.dart in session sodff
```
So I don't think this has anything to do with the debugger, more about how we actually load and run stuff when we have a direct connection to the fletch-vm on the device/qemu and use the fletch command to execute/debug things there.

ricowind, 2015-12-18:
Looking a bit further into this:
This has nothing to do with the debugger or ffi, the code is actually run on a vm on the local machine.

You can validate this by applying:
```
diff --git a/makefile b/makefile
index d31f73e..6fb22df 100644
--- a/makefile
+++ b/makefile
@@ -58,7 +58,7 @@ fletch-session: fletch-reset
 
 fletch-debug-qemu: fletch-session
        $(FLETCH_TOOL_DIR)fletch attach in session sodff tcp_socket 0.0.0.0:4567
-       $(FLETCH_TOOL_DIR)fletch debug ${ARGS} in session sodff
+#      $(FLETCH_TOOL_DIR)fletch run ${ARGS} in session sodff
 
 %.snap: %.dart fletch-session
        $(FLETCH_TOOL_DIR)fletch export $< to $@ in session sodff
```
Now do everything as you did it.

After that, kill the qemu
Locally now run:
```
third_party/fletch/out/ReleaseIA32/fletch debug dart/examples/buttons/buttons_sync.dart in session sodff
```
So, basically, the session is not connected to the vm on the device, it just runs it locally.

@wibling @madsager : did we fix anything with connecting to the socket of a remote vm on a device, and is the connect command here actually correct: https://github.com/domokit/sod/blob/master/makefile#L60

ricowind, 2015-12-18:
Same thing happens if I roll to TOT fletch in the DEPS

ricowind, 2015-12-18:
The log from after running:
```
$ third_party/fletch/out/ReleaseIA32/fletch show log in session sodff
[sodff: 2015-12-18 10:08:33.736 created session with file:///usr/local/google/home/ricow/src/sod/sod/dart/fletch-settings Settings(packages: file:///usr/local/google/home/ricow/src/sod/sod/dart/.packages, options: [--verbose], constants: {}, device_address: null, device_type: null, incremental_mode: IncrementalMode.none)]
[sodff: 2015-12-18 10:08:33.775 Connected to TCP vmSocket  vmSocket 50043 -> 127.0.0.1:4567]
[sodff: 2015-12-18 10:08:46.332 Compiled 'file:///usr/local/google/home/ricow/src/sod/sod/dart/examples/buttons/buttons_sync.dart' to 4075 commands]
[sodff: 2015-12-18 10:08:46.332 Cannot reuse existing VM session, creating new.]
[sodff: 2015-12-18 10:08:46.359 Connected to TCP vmSocket  vmSocket 45124 -> 127.0.0.1:41132]
[sodff: 2015-12-18 10:08:46.709 Uncaught error]
```
So we somehow decide that we can't use that connection and spawn a new vm locally

wibling, 2015-12-18:
Do we have the output from the fletch attach command? Could the specified
port be incorrect? I.e. in order to use the same port each time the
fletch-vm would have to be started with the --port=4567 flag.

/gustav

ricowind, 2015-12-18:
This turns out to be a bug in the fletch driver, introduced by fixing another bug where we would hit an assertion on doing run in a session with an existing failure.
Gustav will file individual bugs.

Temporary workaround for sod:
https://codereview.chromium.org/1539873002/

cpu-chromium, 2015-12-18:
workaround works, thanks!

Let me know if you want me to land that.

ricowind, 2015-12-21:
Carlos: Yes, please land it, I don't have commit access to the SOD repo

wibling, 2015-12-21:
Fixed as of 2cdfd792ffe62d0dda3da62ae226324c07199327.

closed 2015-12-21 by wibling