dart-archive/sdk #385: [sod]   vm/port.cc:37: error: expected: ref_count_ > 0

cpu-chromium, 2015-12-18:
From cousin project SoD.

The change up for review here adds timers via LK ports
https://codereview.chromium.org/1536843004/

When run, it causes an assert in vm/port.cc to fail here:

oid Port::IncrementRef() {
  ASSERT(ref_count_ > 0);       <---- here
  ref_count_++;
}

I have commented out the obvious decrement in that file and removed the assert but then the other assert fails:

void Port::DecrementRef() {
  Lock();
  ASSERT(ref_count_ > 0);    <----- here
  if (--ref_count_ == 0) { ...}
  ....
}

The issue happens after the timer fires twice, I am guessing in the process or reading the 3rd packet from the port. Here is the output of  make qemu-run:

fletch debug
starting fletch-vm...
ready for fletch debug via port: 4567
] 0
= on timer =
0
= on timer =
../../src/vm/port.cc:37: error: expected: ref_count_ > 0
Aborted (immediate)

Note that to use fletch debug you should look at the instructions and the fix in  https://github.com/dart-lang/fletch/issues/384

madsager, 2015-12-21:
Anders, are we seeing this with our timer implementation as well or are we readding the port to the event handler on each event to get balanced ref/deref?

mkustermann, 2015-12-21:
IIRC the eventhandler treats all events as one-shot-events (i.e. you give it a port and it notifies you **at most once** about an event). Every time the event handler sends an event to dart, it will deref the port, see [here](https://github.com/dart-lang/fletch/blob/master/src/vm/event_handler.cc#L132).

skabet, 2015-12-22:
Fixed this yesterday.

closed 2015-12-22 by skabet

mkustermann, 2015-12-22:
@skabet Your change seems to be just not decrementing the refcount on [Port] when sending messages on LK. Are we ever freeing these [Port] objects now or are they leaked?

skabet, 2015-12-22:
We do not have support to stop listening on LK, at this point. A planned
refactor in these areas, should address this.