dart-archive/sdk #387: [sod] fletch compilation broken

cpu-chromium, 2015-12-21:
Did a local roll of fletch

diff --git a/DEPS b/DEPS
   # Fletch repo and dependencies.
-  "fletch_rev": "@193aea1ca0f536c90997baf24b1fee4ab8c9c1b1",
+  "fletch_rev": "@027834ade785623896ba57144157618630b1afae",

Now fletch export is broken.

work/src/fnlk/sod$ make dart/examples/text/text.snap

ninja -C third_party/fletch/ && ninja -C third_party/fletch/out/ReleaseIA32/
ninja: Entering directory `third_party/fletch/'
ninja: no work to do.
ninja: Entering directory `third_party/fletch/out/ReleaseIA32/'
ninja: no work to do.
third_party/fletch/out/ReleaseIA32/fletch quit
Background process exited
third_party/fletch/out/ReleaseIA32/fletch create session sodff with file dart/fletch-settings
Info: Using library root: file:///work/src/fnlk/sod/third_party/fletch/lib/
Info: Using package config: file:///work/src/fnlk/sod/dart/.packages
third_party/fletch/out/ReleaseIA32/fletch export dart/examples/text/text.dart to dart/examples/text/text.snap in session sodff
Info: Using library root: file:///work/src/fnlk/sod/third_party/fletch/lib/
Info: Using package config: file:///work/src/fnlk/sod/dart/.packages
Info: Using platform configuration at file:///work/src/fnlk/sod/third_party/fletch/lib/fletch_mobile.platform
Info: Compiling file:///work/src/fnlk/sod/dart/examples/text/text.dart (build number could not be determined)
Info: Scanning library file:///work/src/fnlk/sod/dart/examples/text/text.dart
Info: Scanning library dart:core (file:///work/src/fnlk/sod/third_party/fletch/third_party/dart/sdk/lib/core/core.dart)
Info: Scanning library dart:fletch._system (file:///work/src/fnlk/sod/third_party/fletch/lib/system/system.dart)
Info: Scanning library dart:_internal (file:///work/src/fnlk/sod/third_party/fletch/third_party/dart/sdk/lib/internal/internal.dart)
Info: Scanning library dart:collection (file:///work/src/fnlk/sod/third_party/fletch/third_party/dart/sdk/lib/collection/collection.dart)
Info: Scanning library dart:math (file:///work/src/fnlk/sod/third_party/fletch/third_party/dart/sdk/lib/math/math.dart)
Info: Scanning library dart:fletch (file:///work/src/fnlk/sod/third_party/fletch/lib/fletch/fletch.dart)
Info: Scanning library dart:convert (file:///work/src/fnlk/sod/third_party/fletch/third_party/dart/sdk/lib/convert/convert.dart)
Info: Scanning library dart:async (file:///work/src/fnlk/sod/third_party/fletch/third_party/dart/sdk/lib/async/async.dart)
Info: Scanning library dart:fletch.os (file:///work/src/fnlk/sod/third_party/fletch/lib/os/os.dart)
Info: Scanning library dart:fletch.ffi (file:///work/src/fnlk/sod/third_party/fletch/lib/ffi/ffi.dart)
Info: Scanning library dart:typed_data (file:///work/src/fnlk/sod/third_party/fletch/third_party/dart/sdk/lib/typed_data/typed_data.dart)
Info: Scanning library package:lk/framebuffer.dart (file:///work/src/fnlk/sod/dart/pkg/lk/lib/framebuffer.dart)
Info: Scanning library package:lk/font.dart (file:///work/src/fnlk/sod/dart/pkg/lk/lib/font.dart)
Info: Resolving...
Info: Resolved 469 elements.
Info: Inferring types...
Info: Compiling...
dart/examples/text/text.dart:9:1:
Internal Error: The compiler crashed when compiling this element.
main () {
^^^^^^^
The compiler is broken.

When compiling the above element, the compiler crashed. It is not
possible to tell if this is caused by a problem in your program or
not. Regardless, the compiler should not crash.

The Dart team would greatly appreciate if you would take a moment to
report this problem at http://dartbug.com/new.

Please include the following information:

* the name and version of your operating system,

* the Dart SDK build number (build number could not be determined), and

* the entire message you see here (including the full stack trace
  below as well as the source location above).

The Fletch program is broken and has crashed.

The Fletch team would greatly appreciate if you would take a moment to
report this problem at https://github.com/dart-lang/fletch/issues/new

Please include the following information:

* the name and version of your operating system

* the Fletch SDK version (0.2.0-edge.027834ade785623896ba57144157618630b1afae)

* the entire message you see here (including the full stack trace below)
Class '_CompactLinkedHashSet<Element>' has no instance method '[]'.

NoSuchMethodError: method not found: '[]'
Receiver: Instance of '_CompactLinkedHashSet<Element>'
Arguments: [Instance of 'PartialMethodElement']
#0      Object._noSuchMethod (dart:core-patch/object_patch.dart:42)
#1      Object.noSuchMethod (dart:core-patch/object_patch.dart:45)
#2      CodegenVisitor.checkCompileError (package:fletchc/src/codegen_visitor.dart:3055:55)
#3      FunctionCodegen.compile (package:fletchc/src/function_codegen.dart:47:9)
#4      FletchBackend.codegenFunction (package:fletchc/src/fletch_backend.dart:974:15)
#5      FletchBackend.compileElement.<anonymous closure> (package:fletchc/src/fletch_backend.dart:802:9)
#6      _CompilerDiagnosticReporter.withCurrentElement (package:compiler/src/compiler.dart:1710:15)
#7      FletchDiagnosticReporter.withCurrentElement (package:fletchc/src/fletch_compiler_implementation.dart:333:30)
#8      FletchBackend.compileElement (package:fletchc/src/fletch_backend.dart:791:23)
#9      FletchEnqueuer.processQueue (package:fletchc/src/fletch_enqueuer.dart:219:40)
#10     FletchEnqueuer.forEach (package:fletchc/src/fletch_enqueuer.dart:199:5)
#11     Compiler.emptyQueue (package:compiler/src/compiler.dart:1106:11)
#12     Compiler.processQueue (package:compiler/src/compiler.dart:1134:5)
#13     Compiler.compileLoadedLibraries (package:compiler/src/compiler.dart:1057:5)
#14     Compiler.runInternal.<anonymous closure> (package:compiler/src/compiler.dart:877:7)
#15     _rootRunUnary (dart:async/zone.dart:914)
#16     _CustomZone.runUnary (dart:async/zone.dart:810)
#17     _Future._propagateToListeners.handleValueCallback (dart:async/future_impl.dart:502)
#18     _Future._propagateToListeners (dart:async/future_impl.dart:585)
#19     _Future._completeWithValue (dart:async/future_impl.dart:376)
#20     _Future._asyncComplete.<anonymous closure> (dart:async/future_impl.dart:430)
#21     _rootRun (dart:async/zone.dart:907)
#22     _CustomZone.run (dart:async/zone.dart:802)
#23     _CustomZone.bindCallback.<anonymous closure> (dart:async/zone.dart:735)
#24     _microtaskLoop (dart:async/schedule_microtask.dart:43)
#25     _microtaskLoopEntry (dart:async/schedule_microtask.dart:52)
#26     _runPendingImmediateCallback (dart:isolate-patch/isolate_patch.dart:96)
#27     _RawReceivePortImpl._handleMessage (dart:isolate-patch/isolate_patch.dart:149)

make: *** [dart/examples/text/text.snap] Error 253

cpu-chromium, 2015-12-21:
I tried to update the revision of dart as well with no success. In the CL I have pasted the other errors

https://codereview.chromium.org/1542563003/

ricowind, 2015-12-22:
Carlos: I can't repro this error, it compiles fine in my checkout after applying your patch from:
https://codereview.chromium.org/1542563003/

Did you by any change forget to run gclient sync after updating the dart_rev and did that go as expected (you can validate by checking that third_party/fletch/third_party/dart is at a5163f3595ccc4f41147a552c31c33090fe3970b and that there is no local changes)

Output from my checkout:
```
nobbo:sod ricow (fixit)$ git diff origin/master
diff --git a/DEPS b/DEPS
index 6edb168..0f1260a 100644
--- a/DEPS
+++ b/DEPS
@@ -16,9 +16,9 @@ vars = {
   "lk_rev": "@6b208387f5e44fdf19b47dd49cea5c153aadc270",
 
   # Fletch repo and dependencies.
-  "fletch_rev": "@193aea1ca0f536c90997baf24b1fee4ab8c9c1b1",
-  "gyp_rev": "@6ee91ad8659871916f9aa840d42e1513befdf638",
-  "dart_rev": "@34357cdad108dcba734949bd13bd28c76ea285e0",
+  "fletch_rev": "@027834ade785623896ba57144157618630b1afae",
+  "gyp_rev": "@6fb8bd829f0ca8fd432fd85ede788b6881c4f09f",
+  "dart_rev": "@a5163f3595ccc4f41147a552c31c33090fe3970b",
   "persistent_rev": "@55daae1a038188c49e36a64e7ef132c4861da3d8",
   "charcode_tag": "@1.1.0",
   "path_tag": "@1.3.6",
nobbo:sod ricow (fixit)$ cd third_party/fletch/
nobbo:fletch ricow (HEAD)$ git log -n1
commit 027834ade785623896ba57144157618630b1afae
Author: Stephan Herhut <herhut@google.com>
Date:   Mon Dec 21 16:15:46 2015 +0100

    Actually use global library in ffi test.
    
    TBR=ricow@google.com
    BUG=
    
    Review URL: https://codereview.chromium.org/1542643002 .
nobbo:fletch ricow (HEAD)$ cd third_party/dart
nobbo:dart ricow (HEAD)$ git log -n1
commit a5163f3595ccc4f41147a552c31c33090fe3970b
Author: Sigurd Meldgaard <sigurdm@google.com>
Date:   Tue Dec 15 10:56:45 2015 +0100

    Associate compile-time errors with elements
    BUG=
    R=johnniwinther@google.com
    
    Review URL: https://codereview.chromium.org/1525603002 .
nobbo:dart ricow (HEAD)$ cd ../../../..
nobbo:sod ricow (fixit)$ make dart/examples/text/text.snap
ninja -C third_party/fletch/ && ninja -C third_party/fletch/out/ReleaseIA32/
ninja: Entering directory `third_party/fletch/'
ninja: no work to do.
ninja: Entering directory `third_party/fletch/out/ReleaseIA32/'
ninja: no work to do.
third_party/fletch/out/ReleaseIA32/fletch quit
Background process exited
third_party/fletch/out/ReleaseIA32/fletch create session sodff with file dart/fletch-settings
Info: Using library root: file:///usr/local/google/home/ricow/src/sod/sod/third_party/fletch/lib/
Info: Using package config: file:///usr/local/google/home/ricow/src/sod/sod/dart/.packages
third_party/fletch/out/ReleaseIA32/fletch export dart/examples/text/text.dart to dart/examples/text/text.snap in session sodff
Info: Using library root: file:///usr/local/google/home/ricow/src/sod/sod/third_party/fletch/lib/
Info: Using package config: file:///usr/local/google/home/ricow/src/sod/sod/dart/.packages
Info: Using platform configuration at file:///usr/local/google/home/ricow/src/sod/sod/third_party/fletch/lib/fletch_mobile.platform
Info: Compiling file:///usr/local/google/home/ricow/src/sod/sod/dart/examples/text/text.dart (build number could not be determined)
Info: Scanning library file:///usr/local/google/home/ricow/src/sod/sod/dart/examples/text/text.dart
Info: Scanning library dart:core (file:///usr/local/google/home/ricow/src/sod/sod/third_party/fletch/third_party/dart/sdk/lib/core/core.dart)
Info: Scanning library dart:fletch._system (file:///usr/local/google/home/ricow/src/sod/sod/third_party/fletch/lib/system/system.dart)
Info: Scanning library dart:_internal (file:///usr/local/google/home/ricow/src/sod/sod/third_party/fletch/third_party/dart/sdk/lib/internal/internal.dart)
Info: Scanning library dart:collection (file:///usr/local/google/home/ricow/src/sod/sod/third_party/fletch/third_party/dart/sdk/lib/collection/collection.dart)
Info: Scanning library dart:math (file:///usr/local/google/home/ricow/src/sod/sod/third_party/fletch/third_party/dart/sdk/lib/math/math.dart)
Info: Scanning library dart:fletch (file:///usr/local/google/home/ricow/src/sod/sod/third_party/fletch/lib/fletch/fletch.dart)
Info: Scanning library dart:convert (file:///usr/local/google/home/ricow/src/sod/sod/third_party/fletch/third_party/dart/sdk/lib/convert/convert.dart)
Info: Scanning library dart:async (file:///usr/local/google/home/ricow/src/sod/sod/third_party/fletch/third_party/dart/sdk/lib/async/async.dart)
Info: Scanning library dart:fletch.os (file:///usr/local/google/home/ricow/src/sod/sod/third_party/fletch/lib/os/os.dart)
Info: Scanning library dart:fletch.ffi (file:///usr/local/google/home/ricow/src/sod/sod/third_party/fletch/lib/ffi/ffi.dart)
Info: Scanning library dart:typed_data (file:///usr/local/google/home/ricow/src/sod/sod/third_party/fletch/third_party/dart/sdk/lib/typed_data/typed_data.dart)
Info: Scanning library package:lk/framebuffer.dart (file:///usr/local/google/home/ricow/src/sod/sod/dart/pkg/lk/lib/framebuffer.dart)
Info: Scanning library package:lk/font.dart (file:///usr/local/google/home/ricow/src/sod/sod/dart/pkg/lk/lib/font.dart)
Info: Resolving...
Info: Resolved 469 elements.
Info: Inferring types...
Info: Compiling...
Info: Total compile-time 2093msec; unaccounted 2093msec
nobbo:sod ricow (fixit)$ 

```

peter-ahe-google, 2016-01-04:
I'm going to close this as I'm pretty sure I saw this exact problem when forgetting to run `gclient sync` a few weeks ago. Please reopen if this is still reproducible after running `gclient sync`.

closed 2016-01-04 by peter-ahe-google