dart-archive/sdk #388: Create packages for Uri and Regexp

sigurdm, 2015-12-22:
For the embedded core libraries we have removed Uri and Regexp classes (https://github.com/dart-lang/fletch/issues/176).
We should make packages that include replacements of these classes available.

mit-mit, 2015-12-29:
For RegExp, maybe we should do something minimal, e.g. something like [SLRE](https://github.com/cesanta/slre). 

@ErikCorryGoogle, wdyt?

ErikCorryGoogle, 2016-01-04:
I think it's not worth bothering with SLRE unless someone is actually using regexps on Fletch and reports them to be too big or too slow.  The current solution is tree-shaken out, so I'm not even sure there was anything to be gained by removing regexp from the core libraries.

sgjesse, 2016-01-04:
The core libraries should only contain APIs which are core, and should be
expected to "work well" on the given platform. I don't think this is the
case for RegExp, and moving it to a package clearly indicate this. It also
makes it clear that RegExp cannot be used by other dart: libraries.

With RegExp in a package people are free to use it, or use other RegExp
packages (e.g. based in SLRE) if available. This can of cause have the side
effect that transitively one can end up with several RegExp implementations.

ErikCorryGoogle, 2016-01-04:
With RegExp in core, people are still free to use a different RegExp implementation, and the core RegExp can still be shaken out.  I fail to see the gain here.  And we are still free to not use RegExp from other dart: libraries.