dart-archive/sdk #391: calling a getter on a bytearray with () will make us say that _Smi has no method named call

ricowind, 2016-01-07:
@peter-ahe-google @karlklose @sigurdm 

OS: linux
Fletch commit: 55c322daadb65f90615c7230391814153a43bd70

How to reproduce the problem:
```
import 'dart:typed_data';

void main() {
  var buffer = new Uint8List(42).buffer;
  buffer.lengthInBytes();
}
```

Expected result: Stating that I can't do a call on an integer

Actual result:
```
~/src/fletch/fletch/out/DebugX64/fletch run buffer.dart
Uncaught exception: NoSuchMethodError: Class '_Smi' has no method named 'call' that takes 0 arguments (3::45056)
Stack trace:
> 0: main                            @buffer.lengthInBytes()

buffer.dart:5:3
1    import 'dart:typed_data';
2    
3    void main() {
4      var buffer = new Uint8List(42).buffer;
5      buffer.lengthInBytes();
6    }
```

peter-ahe-google, 2016-01-07:
As far as I can tell, you're getting told that you can't do a call (no method named 'call') on an integer (Class '_Smi').

Could you clarify what you think should happen and how that's different from the observed result?

ricowind, 2016-01-07:
I am not being told I can't do a call on an integer, that is what I wan't to be told!  I am being told that I can't do call on _Smi.
_I_ know what a smi is, I don't expect a non dart team developer to know what a smi is though