dart-archive/sdk #394: DebugInfo computation fails after trivial incremental compilation

madsager, 2016-01-14:
Dart program:

```dart
import 'dart:fletch';

void noop() {}

void main() {
  Process.spawnDetached(noop);
  throw 42;
}
```

Running the program twice leads to an error during debug info computation:

```
$ ./out/ReleaseX64ClangAsan/fletch run test.dart
Uncaught exception: 42
Stack trace:
> 0: main        @throw 42

test.dart:7:3
2    
3    void noop() {}
4    
5    void main() {
6      Process.spawnDetached(noop);
7      throw 42;

$ ./out/ReleaseX64ClangAsan/fletch run test.dart
Uncaught exception: 42
file:///usr/local/google/home/ager/ssd/fletch-repo/fletch/test.dart+39-50: The compiler crashed when compiling this element.
The compiler is broken.

When compiling the above element, the compiler crashed. It is not
possible to tell if this is caused by a problem in your program or
not. Regardless, the compiler should not crash.

The Dart team would greatly appreciate if you would take a moment to
report this problem at http://dartbug.com/new.

Please include the following information:

* the name and version of your operating system,

* the Dart SDK build number (build number could not be determined), and

* the entire message you see here (including the full stack trace
  below as well as the source location above).

type 'FletchFunction' is not a subtype of type 'FletchFunctionBuilder' of 'functionBuilder'.
#0      FletchBackend.createTearoffClass (package:fletchc/src/fletch_backend.dart:528:25)
#1      CodegenVisitor.handleStaticFunctionGet (package:fletchc/src/codegen_visitor.dart:1011:25)
#2      SemanticVisitor&&GetBulkMixin&&SetBulkMixin&&ErrorBulkMixin&&InvokeBulkMixin&&IndexSetBulkMixin&&CompoundBulkMixin&&UnaryBulkMixin&&BaseBulkMixin&&BinaryBulkMixin&&PrefixBulkMixin&&PostfixBulkMixin&&NewBulkMixin&&VariableBulkMixin&&ParameterBulkMixin&&FunctionBulkMixin&&ConstructorBulkMixin&&InitializerBulkMixin&&BaseImplementationOfStaticsMixin.visitTopLevelFunctionGet (package:compiler/src/resolution/semantic_visitor_mixins.dart:7895:12)
#3      GetStructure.dispatch (package:compiler/src/resolution/send_structure.dart:493:24)
#4      Visitor&SemanticSendResolvedMixin.visitSend (package:compiler/src/resolution/semantic_visitor.dart:54:24)
#5      Send.accept (package:compiler/src/tree/nodes.dart:383:38)
#6      CodegenVisitor.doVisitForValue (package:fletchc/src/codegen_visitor.dart:572:10)
#7      CodegenVisitor.visitForValue (package:fletchc/src/codegen_visitor.dart:578:5)
#8      DebugInfoFunctionCodegen.visitForValue (package:fletchc/src/debug_info_function_codegen.dart:146:24)
#9      CodegenVisitor.doStaticFunctionInvoke (package:fletchc/src/codegen_visitor.dart:505:9)
#10     CodegenVisitor.doStaticallyBoundInvoke (package:fletchc/src/codegen_visitor.dart:1082:5)
#11     CodegenVisitor.handleStaticFunctionInvoke (package:fletchc/src/codegen_visitor.dart:1091:5)
#12     SemanticVisitor&&GetBulkMixin&&SetBulkMixin&&ErrorBulkMixin&&InvokeBulkMixin&&IndexSetBulkMixin&&CompoundBulkMixin&&UnaryBulkMixin&&BaseBulkMixin&&BinaryBulkMixin&&PrefixBulkMixin&&PostfixBulkMixin&&NewBulkMixin&&VariableBulkMixin&&ParameterBulkMixin&&FunctionBulkMixin&&ConstructorBulkMixin&&InitializerBulkMixin&&BaseImplementationOfStaticsMixin.visitStaticFunctionInvoke (package:compiler/src/resolution/semantic_visitor_mixins.dart:7820:12)
#13     InvokeStructure.dispatch (package:compiler/src/resolution/send_structure.dart:207:24)
#14     Visitor&SemanticSendResolvedMixin.visitSend (package:compiler/src/resolution/semantic_visitor.dart:54:24)
#15     Send.accept (package:compiler/src/tree/nodes.dart:383:38)
#16     CodegenVisitor.visitForEffect (package:fletchc/src/codegen_visitor.dart:595:10)
#17     DebugInfoFunctionCodegen.visitForEffect (package:fletchc/src/debug_info_function_codegen.dart:151:25)
#18     CodegenVisitor.visitExpressionStatement (package:fletchc/src/codegen_visitor.dart:2567:5)
#19     ExpressionStatement.accept (package:compiler/src/tree/nodes.dart:1164:38)
#20     CodegenVisitor.doStatements (package:fletchc/src/codegen_visitor.dart:2537:17)
#21     CodegenVisitor.visitBlock (package:fletchc/src/codegen_visitor.dart:2559:5)
#22     Block.accept (package:compiler/src/tree/nodes.dart:617:38)
#23     FunctionCodegen.compile (package:fletchc/src/function_codegen.dart:110:17)
#24     FletchBackend.createDebugInfo.<anonymous closure> (package:fletchc/src/fletch_backend.dart:769:66)
#25     _CompilerDiagnosticReporter.withCurrentElement (package:compiler/src/compiler.dart:1727:15)
#26     FletchDiagnosticReporter.withCurrentElement (package:fletchc/src/fletch_compiler_implementation.dart:333:30)
#27     FletchBackend.createDebugInfo (package:fletchc/src/fletch_backend.dart:769:25)
#28     IncrementalCompiler.createDebugInfo (package:fletchc/incremental/fletchc_incremental.dart:295:38)
#29     DebugState.getDebugInfo.<anonymous closure> (package:fletchc/debug_state.dart:109:31)
#30     _HashVMBase&MapMixin&&_LinkedHashMapMixin.putIfAbsent (dart:collection-patch/compact_hash.dart:254)
#31     DebugState.getDebugInfo (package:fletchc/debug_state.dart:108:23)
#32     BackTraceFrame.debugInfo (package:fletchc/back_trace.dart:31:41)
#33     BackTraceFrame.sourceLocation (package:fletchc/back_trace.dart:74:12)
#34     BackTrace.sourceLocation (package:fletchc/back_trace.dart:196:22)
#35     DebugState.currentBackTrace= (package:fletchc/debug_state.dart:95:34)
#36     Session.backTrace.<backTrace_async_body> (package:fletchc/vm_session.dart:579:18)
#37     _asyncThenWrapperHelper.<anonymous closure> (dart:async-patch/async_patch.dart:27)
#38     _rootRunUnary (dart:async/zone.dart:914)
#39     _CustomZone.runUnary (dart:async/zone.dart:810)
#40     _Future._propagateToListeners.handleValueCallback (dart:async/future_impl.dart:502)
#41     _Future._propagateToListeners (dart:async/future_impl.dart:585)
#42     _Future._complete (dart:async/future_impl.dart:366)
#43     _SyncCompleter.complete (dart:async/future_impl.dart:52)
#44     FletchVmSession.runCommands.<runCommands_async_body> (package:fletchc/vm_session.dart)
#45     _asyncThenWrapperHelper.<anonymous closure> (dart:async-patch/async_patch.dart:27)
#46     _rootRunUnary (dart:async/zone.dart:914)
#47     _CustomZone.runUnary (dart:async/zone.dart:810)
#48     _Future._propagateToListeners.handleValueCallback (dart:async/future_impl.dart:502)
#49     _Future._propagateToListeners (dart:async/future_impl.dart:585)
#50     _Future._complete (dart:async/future_impl.dart:366)
#51     _SyncCompleter.complete (dart:async/future_impl.dart:52)
#52     FletchVmSession.readNextCommand.<readNextCommand_async_body> (package:fletchc/vm_session.dart)
#53     _asyncThenWrapperHelper.<anonymous closure> (dart:async-patch/async_patch.dart:27)
#54     _rootRunUnary (dart:async/zone.dart:914)
#55     _CustomZone.runUnary (dart:async/zone.dart:810)
#56     _Future._propagateToListeners.handleValueCallback (dart:async/future_impl.dart:502)
#57     _Future._propagateToListeners (dart:async/future_impl.dart:585)
#58     _Future._complete (dart:async/future_impl.dart:366)
#59     _StreamIteratorImpl._onData (dart:async/stream_impl.dart:1051)
#60     _rootRunUnary (dart:async/zone.dart:914)
#61     _CustomZone.runUnary (dart:async/zone.dart:810)
#62     _CustomZone.runUnaryGuarded (dart:async/zone.dart:716)
#63     _BufferingStreamSubscription._sendData (dart:async/stream_impl.dart:341)
#64     _DelayedData.perform (dart:async/stream_impl.dart:595)
#65     _StreamImplEvents.handleNext (dart:async/stream_impl.dart:711)
#66     _PendingEvents.schedule.<anonymous closure> (dart:async/stream_impl.dart:671)
#67     _rootRun (dart:async/zone.dart:907)
#68     _CustomZone.run (dart:async/zone.dart:802)
#69     _CustomZone.bindCallback.<anonymous closure> (dart:async/zone.dart:735)
#70     _microtaskLoop (dart:async/schedule_microtask.dart:43)
#71     _microtaskLoopEntry (dart:async/schedule_microtask.dart:52)
#72     _runPendingImmediateCallback (dart:isolate-patch/isolate_patch.dart:96)
#73     _RawReceivePortImpl._handleMessage (dart:isolate-patch/isolate_patch.dart:149)
```

madsager, 2016-01-14:
@zerny @peter-ahe-google

peter-ahe-google, 2016-01-18:
Proposed fix in [CL 1586123005](https://codereview.chromium.org/1586123005).

commit d7fa4a8f (2016-01-18): Create class builder for tear-off classes from predecessor system

Fixes #394

R=ajohnsen@google.com, zerny@google.com

Review URL: https://codereview.chromium.org/1586123005 .
closed 2016-01-18 by peter-ahe-google