dart-archive/sdk #399: Default name for SD image for pi, 'fletch', causes confusion

lukechurch, 2016-01-25:
The default behaviour of the flash tool is to create an image called 'fletch'. 

This means that the device appears with this name, not the name 'raspberrypi' as appears in the instructions, in e.g. 

```
Hello from Linux running on raspberrypi.
```

I think that in this case it's actually confusing calling the image fletch, as it further overloads the name.

E.g. See the below interaction

```

Looking for Fletch capable devices (will search for 5 seconds)...
1: Device at 192.168.0.31     fletch.local

1 Fletch capable device was found.
Please enter the number or the IP address of the remote device (press enter for the first device): 
Created settings file 'file:///Users/lukechurch/remote.fletch-settings'
Hello from Linux running on fletch.
Lukes-MacBook-Pro-2:fletch-sdk lukechurch$ 

====

```

'fletch.local' and 'Hello from Linux running on fletch' are both confusing.

Consider changing the default name of the image to something like pi-with-fletch

The docs and the tool would need the change.

@mit-mit @sgjesse

closed 2016-02-12 by sgjesse

lukechurch, 2016-02-12:
@sgjesse @mit-mit The docs also need updating to reflect this: https://dartino.org/

I'm guessing we have that already planned for before the next release? I didn't see a tracking bug anywhere?

mit-mit, 2016-02-12:
Yup, tracking doc changes in https://github.com/dartino/sdk/issues/399,
added this

commit a4f31dc4 (2016-02-12): Change the default hostname set when flashing a Raspberry Pi image

Closes #399

BUG= https://github.com/dartino/sdk/issues/399
R=lukechurch@google.com, mit@google.com

Review URL: https://codereview.chromium.org/1686393003 .