dart-archive/sdk #401: Unexplained flaky timeouts on Mac

ErikCorryGoogle, 2016-01-26:
They look like an issue with the driver (VM is idle in core dumps), but only seem to hit long-running tests.

peter-ahe-google, 2016-01-27:
Sounds like the VM isn't shutting down.

mkustermann, 2016-01-27:
@zerny  Could this have been fixed by 18f9665da28d43876608a40589a92dfc756126b7 ?

zerny, 2016-01-27:
Not sure as it does not figure what tests and when they were observed to flake. The timeouts caused by 18f9665 were seen on some of the linux bots too.

ricowind, 2016-02-18:
There are now core dumps of this on the bots, e.g.,
https://uberchromegw.corp.google.com/i/client.fletch/builders/dartino-mac-debug-asan-x86/builds/583
They will no longer be surfaced as red bots since I changed the suppression to Pass, Slow, Crash instead of Pass, Slow, Timeout