dart-archive/sdk #402: Rename art-dependencies-fletch bucket

mit-mit, 2016-01-27:
As part of the fletch->dartino rename we should update the cloud storage bucket we use in DEPS:
```
dart-dependencies-fletch
```

sgjesse, 2016-01-28:
As far as I can see there is no renaming of Cloud Storage buckets, so we will have to create a new one, copy over the content and (maybe) delete the old one.

madsager, 2016-01-28:
The SoD repo also pulls from our cloud storage bucket. So, before deleting the old one we need to make sure to fix their DEPS. I'll be happy to do that.

ricowind, 2016-02-03:
we should never delete that bucket, it will destroy our ability to do old builds

ricowind, 2016-02-03:
I will create a new bucket for our dependencies, copy the existing used ones over and update the deps

mit-mit, 2016-02-03:
OK, that sounds great.

ricowind, 2016-02-04:
done, we now use dartino-dependencies

closed 2016-02-04 by ricowind