dart-archive/sdk #419: backgroundColor & foregroundColor functions in pkg/lcd causes NoSuchMethod error

mit-mit, 2016-02-04:
For example, running this:

```
// Copyright (c) 2015, the Fletch project authors. Please see the AUTHORS file
// for details. All rights reserved. Use of this source code is governed by a
// BSD-style license that can be found in the LICENSE.md file.

import 'package:stm32f746g_disco/stm32f746g_disco.dart';
import 'package:stm32f746g_disco/lcd.dart';

main() {
  // Initialize screen
  STM32F746GDiscovery disco = new STM32F746GDiscovery();
  var display = disco.frameBuffer;

  display.clear(Color.black);
  // display.backgroundColor(Color.black);
  display.foregroundColor(Color.green);

  display.writeText(20,20, 'Hello');
}
```

on the disco board causes a NoSuchMethodError when running the .bin file on the board.

sgjesse, 2016-02-04:
They are setters :-) - maybe that is the issue.

mit-mit, 2016-02-04:
Yup, I'll go hide in the corner now...

closed 2016-02-04 by mit-mit