dart-archive/sdk #426: Type checking of FFI calls

mit-mit, 2016-02-08:
We currently do not do any type checking before making ffi calls. This can cause weird runtime errors when passing the wrong arguments, for example the following program will cause an "uncaught exception" crash as runtime on the STM board:

```
import 'package:stm32f746g_disco/stm32f746g_disco.dart';
import 'package:stm32f746g_disco/lcd.dart';

main() {
  // Initialize display, and get display dimensions.
  STM32F746GDiscovery discoBoard = new STM32F746GDiscovery();
  var display = discoBoard.frameBuffer;

  display.drawLine(200/2, 100, 150, 150, Color.black);
}

```

We should figure what the right solution is here generally. For example, do we want something like "checked mode"?

mit-mit, 2016-05-19:
This should be resolved once we do https://github.com/dartino/sdk/issues/463

mit-mit, 2016-05-20:
Re-opening after further discussion: While static type checking with strong mode will help in cases like this one, it still allows for writing dynamic code where the types will have to be checked at run time, e.g.:

```
main() {
  // Initialize display, and get display dimensions.
  STM32F746GDiscovery discoBoard = new STM32F746GDiscovery();
  var display = discoBoard.frameBuffer;

 int x, y, w;
 dynamic h;
 ...
  display.drawLine(x, y, w, h, Color.black);
}

```

Here the type of h cannot be determined at compile time.

Thus, it would still be very useful to support runtime checking of types before passing them over the dart/native boundary, e.g. having type asserts and a special checked mode in which these are validated.

mit-mit, 2016-07-19:
@DmitryOlshansky-google have you thought about this?

DmitryOlshansky-google, 2016-07-19:
@mit-mit Now we have type-checked FFI and nothing prevents us from using more of it in place of the old FFI interface. Need to compile a list of APIs to convert though.

DmitryOlshansky-google, 2016-08-02:
@mit-mit Should be resolved in commits:
https://github.com/dartino/sdk/commit/908bf02e1add5a2cc5d918a36846cccf28b6d9e9
https://github.com/dartino/sdk/commit/ab9754ded2aed937156fd47fac66e53d39f79986

Now calling with double instead of integer just truncates the double.

closed 2016-08-02 by DmitryOlshansky-google