dart-archive/sdk #427: lcd package: text output at unexpected location

mit-mit, 2016-02-08:
If you clear the screen and then render text in a tight loop, the text occasionally is rendered at the wrong location, for example this code:

```
import 'package:stm32f746g_disco/stm32f746g_disco.dart';
import 'package:stm32f746g_disco/lcd.dart';

main() {
  STM32F746GDiscovery discoBoard = new STM32F746GDiscovery();
  var display = discoBoard.frameBuffer;

  while (true) {
    display.clear(Color.black);
    display.writeText(100, 100, 'HHHEEELLLOOOOO');
  }
}
```

sigurdm, 2016-02-08:
We could try doing the same operations in a tight c-loop.

sgjesse, 2016-02-08:
The foreign call was missing an argument. The last argument was for alignment, and that was getting a "random" value from the stack.

commit 3405a5e3 (2016-02-09): Update the LCD support in the STM32F746G Discovery board package

Added read and write of pixels and drawing of circles.

Fixed text drawing issue where the alignment argument was not passed to
the forign call.

Closes #427

BUG= https://github.com/dartino/sdk/issues/427
R=sigurdm@google.com

Review URL: https://codereview.chromium.org/1681633002 .
closed 2016-02-09 by sgjesse