dart-archive/sdk #429: knightrider sample does not compile

mit-mit, 2016-02-15:
Running `./bin/dartino flash samples/stm32f746g-discovery/knight-rider.dart` in 0.3.0-dev.4.0 produces a lot of compilation errors:

```
mit-macbookpro:dartino-sdk-0.3.0-dev.4.0 mit$ ./bin/dartino flash samples/stm32f746g-discovery/knight-rider.dart
samples/stm32f746g-discovery/knight-rider.dart:27:8:
Warning: Cannot resolve type 'Pin'.
  List<Pin> leds = [
       ^^^
samples/stm32f746g-discovery/knight-rider.dart:46:9:
Warning: Cannot resolve type 'Gpio'.
  final Gpio _gpio;
        ^^^^

...
```

sgjesse, 2016-02-16:
I cannot reproduce this. Running in a freshly downloaded Dartino SDK 0.3.0-dev.4.0:

```
$  bin/dartino --version
0.3.0-dev.4.0
$ bin/dartino x-download-tools
$ bin/dartino flash samples/stm32f746g-discovery/knight-rider.dart 
Done building image: /tmp/dartino-sdk-0.3.0-dev.4.0/samples/stm32f746g-discovery/knight-rider.bin
Flashing image: file:///tmp/dartino-sdk-0.3.0-dev.4.0/samples/stm32f746g-discovery/knight-rider.bin
Done flashing image: /tmp/dartino-sdk-0.3.0-dev.4.0/samples/stm32f746g-discovery/knight-rider.bin
```

It seems like it could be related to an old `~/local.fletch-settings`.

The new batch-mode (when not using sessions) still defaults to session "local" settings.

mit-mit, 2016-02-16:
Yes, that fixed it, thanks!

closed 2016-02-16 by mit-mit