dart-archive/sdk #434: language/override_inheritance_no_such_method_test/none hits vm assertion on mac asan

ricowind, 2016-02-18:
@mkustermann @peter-ahe-google 

See e.g.:
https://uberchromegw.corp.google.com/i/client.fletch/builders/dartino-mac-debug-asan-x86/builds/587/steps/test_DebugIA32ClangAsan-snapshot%20failures/logs/stdio

I will mark this as flaky in this config in the status files

zerny, 2016-02-19:
Previously discussed in #401

mkustermann, 2016-03-01:
This is discussing the same issue as #401 - which has more context - so I'll close this CL.

mkustermann, 2016-03-01:
Issue #401 is actually not the same (it is about timeouts and accidentally mentions this issue as well), reopening.

To give an update on this and a quick summary of this issue:

When running tests with `--compiler=dartino_compiler --runtime=dartinovm` our tests will generate snapshots using the `dartino_compiler`. In order to write a snapshot a program will be GCed and compacted. Then the snapshot will be written to a file and the **offsets of Function and Class objects** are collected. These functions and klasses are then converted to **id**s (via the **Function/Class -> id** mappings in `src/vm/session.cc`). Afterwards the VM sends a list of **(klass/function-id, offset)** back to the `dartino_compiler`.

Now sometimes the dartino VM cannot find a function/klass in the **Function/Class -> id** maps. This is an error and should never happen. We then hit an assertion I added to the VM [here](https://github.com/dartino/sdk/blob/292f3e5537a45dd9042952b0528b13ebbc8f4216/src/vm/session.cc#L726):
```
     ../../src/vm/session.cc:689: error: expected: id != -1
```
It is unclear what causes this (in particular: It is very surprising that a) this is flaky b) it happens on specific tests). Two possible reasons are:
  * After doing GCs objects get moved but the keys in the session maps do not get updated.
  * The incremental compiler is not pushing the ids of klasses/functions

I've made a CL which adds an `IdVerifier` to the VM which will ensure that each Function/Klass object **does have an id** associated with it and runs this `IdVerifier` after committing changes. The [CL](https://codereview.chromium.org/1750853003) is currently failing a number of incremental tests (for not-yet-known reasons) and is therefore not committed.

There are currently 2 tests marked as flaky in the status file:
```
tests/language.status:override_inheritance_no_such_method_test/none: Pass, Crash # Issue 434
tests/language.status:setter_override_test/none: Pass, Crash # Issue 434
```

I probably won't have time to investigate this further, so I'll leave this up to someone else.

cc interested parties:
@ricowind 
@peter-ahe-google (for incremental compiler)
@ErikCorryGoogle (for gc)