dart-archive/sdk #443: Initial `fetch dartino` takes too long

floitschG, 2016-02-22:
I just tried to follow the build instructions for dartino.
At the `fetch dartino` step, the process seems to get stuck. In reality it's (probably?) downloading something stm related in the background.
This is unexpected:
- it takes forever,
- the build instructions should warn that a huge amount of space will be needed.

Alternatively, the stm things should be optional.

mit-mit, 2016-05-19:
I don't think STM parts should be optional. But it does take a long time. @ricowind do you happen to know why?

ricowind, 2016-05-19:
fetch is part of depot_tools, which is managed by chrome infrastructure, we just added a recipe. You can try doing the same for chromium (fetch chromium), which will take way longer.
The reason _why_ it takes a long time is obvious, we download a lot and the office internet connection is not that fast.
We could file a bug against depot tools to ask for progress indication when running, but I think we should just update the docs instead:
  -- This will pull down a few gigabytes including a cross compilation toolchain and the stm libraries, so the step will take a long time (without any progress indication).
  -- You can also run fetch --nohooks dartino for a fast clone, but you then manually have to run gclient sync to get the dependencies pulled down afterwards.

mit-mit, 2016-05-19:
Fixed in https://codereview.chromium.org/1989873003/

closed 2016-05-19 by mit-mit