dart-archive/sdk #449: .packages in sdk folder covers both programs implementing and implemented in dartino

sigurdm, 2016-02-26:
sdk/.packages contains a reference to sdk/pkg/dartino
sdk/pkg/dartino contains an embedder.yaml disallowing async.

If you open the sdk in intellij, the newest analyzer will find it, and analyze all of the dartino_compiler etc. in a Dartino context, and underline all async code with red.

This problem is even worse for the dart_agent package that contains code both for running on the VM, and code to run on Dartino. It would have to be split up into two packages (three if there has to be shared code)

We need to set up different package files for Dartino programs, and for the implementation of Dartino (probably a healthy split anyway. Why does a dartino-program need to know about the dartino_compiler package?).

cc @mit-mit @sgjesse

sigurdm, 2016-02-26:
It seems we already have ./pkg/dartino_compiler/.packages, not containing the dartino package. It seems my analyzer did not pick it up.

Will investigate further...

mit-mit, 2016-02-29:
Dan, can you clarify?

sigurdm, 2016-02-29:
Sorry - I think it was me who was confused. The analyzer seems to be doing its job. We just need to put .packages files more places.

sigurdm, 2016-02-29:
I changed the issue title.
But still we have the issue with the dart-agent that has code for both worlds