dart-archive/sdk #453: Template directory names should match agreed upon board names

mit-mit, 2016-03-03:
The final names are:

raspberrypi2
stm32F746Gdiscovery
(and soon: stm32F411nucleo)

sgjesse, 2016-03-04:
I will suggest a naming scheme with all lowercase and underscores for separators

  raspberry_pi2
  stm32f746g_discovery
  stm32f411re_nucleo

and then use the same name for packages and libraries

  pkg/stm32f746g_discovery/lib/stm32f746g_discovery.dart
  pkg/stm32f411re_nucleo/lib/stm32f411re_nucleo.dart

or if we go for a "combined" package

  pkg/stm32/lib/stm32f746g_discovery.dart
  pkg/stm32/lib/stm32f411re_nucleo.dart

and board static libraries - which we will get soon for MCU boards

  libstm32f746g_discovery.a
  libstm32f411re_nucleo.a

mit-mit, 2016-03-04:
I'm not too fond of having underscores, as they are also used as the
'platform target' string that you type in 'dartino create project
<target>'. They are easier to ready, but much harder to type.

Can we use dashes, i.e. stm32f746g-discovery?

sgjesse, 2016-03-04:
For 'dartino create project' we can provide a list if <target> is not given, e.g.

$ dartino create project
Please select the default board for this project:
[1] STM32F746G Discovery (stm32f746g_discovery)
[2] STM32F411RE Nucleo (stm32f411re_nucleo)

I also like dashes better, but the problem with dashes is that they cannot be used in Dart identifier names, but maybe we don't need to have these names in any identifiers. For package imports they should be fine.

In pkg/ we use underbars for all names, and in samples/ we use dashes.

mit-mit, 2016-03-07:
OK, lets go with dashes, and I like seperating the make from the model, i.e.:

```
raspberry-pi2
stm32-f746g-discovery
```
(and soon: `stm32-f411-nucleo`)

sgjesse, 2016-03-07:
I think we should keep the naming as close as possible to the name the
"product", so not add a dash after stm32. I have not seen that done by STM.

danrubel, 2016-03-15:
The platforms have been renamed in the SDK...
```
raspberry-pi2
stm32f746g-discovery
```

danrubel, 2016-03-15:
Cleaned up repo to match SDK structure... https://codereview.chromium.org/1797263002

closed 2016-03-15 by danrubel