dart-archive/sdk #454: `dartino build` fails to build samples\general\tls-socket.dart

mit-mit, 2016-03-04:
On master after gclient sync and ninja'ing everything:

```
mit-pc:sdk mit (master)$ ./out/ReleaseX64Clang/dartino build samples/general/tls-socket.dart 
STDOUT:
Converting snapshot to object file
Linking application

STDERR:
/usr/local/google/home/mit/dev/dartino/sdk/out/ReleaseSTM/libdisco_dartino.a(libdisco_dartino.dartino_entry.o): In function `StartDartino(void const*)':
dartino_entry.cc:(.text._Z12StartDartinoPKv+0x40): undefined reference to `program_start'
dartino_entry.cc:(.text._Z12StartDartinoPKv+0x44): undefined reference to `program_end'
collect2: error: ld returned 1 exit status

Internal error: Failed to build image, error = 1
#0      throwInternalError (package:dartino_compiler/src/diagnostic.dart:206:3)
#1      buildImage.<buildImage_async_body> (package:dartino_compiler/src/worker/developer.dart:1271:7)
#2      _asyncThenWrapperHelper.<anonymous closure> (dart:async-patch/async_patch.dart:27)
#3      _rootRunUnary (dart:async/zone.dart:902)
#4      _CustomZone.runUnary (dart:async/zone.dart:798)
#5      _Future._propagateToListeners.handleValueCallback (dart:async/future_impl.dart:551)
#6      _Future._propagateToListeners (dart:async/future_impl.dart:637)
#7      _Future._completeWithValue (dart:async/future_impl.dart:424)
#8      Future.wait.<anonymous closure> (dart:async/future.dart:299)
#9      _rootRunUnary (dart:async/zone.dart:902)
#10     _CustomZone.runUnary (dart:async/zone.dart:798)
#11     _Future._propagateToListeners.handleValueCallback (dart:async/future_impl.dart:551)
#12     _Future._propagateToListeners (dart:async/future_impl.dart:637)
#13     _Future._completeWithValue (dart:async/future_impl.dart:424)
#14     _Future._asyncComplete.<anonymous closure> (dart:async/future_impl.dart:479)
#15     _rootRun (dart:async/zone.dart:895)
#16     _CustomZone.run (dart:async/zone.dart:790)
#17     _CustomZone.bindCallback.<anonymous closure> (dart:async/zone.dart:723)
#18     _microtaskLoop (dart:async/schedule_microtask.dart:41)
#19     _startMicrotaskLoop (dart:async/schedule_microtask.dart:50)
#20     _runPendingImmediateCallback (dart:isolate-patch/isolate_patch.dart:96)
#21     _RawReceivePortImpl._handleMessage (dart:isolate-patch/isolate_patch.dart:149)

mit-pc:sdk mit (master)$ 

```

karlklose, 2016-03-09:
Looks like this problem started after 49a0acf7a183e4edb38a40cd69e72885bb2f1b0e

sgjesse, 2016-03-09:
The change @karlklose mention changes `libdisco_dartino.a` to expect a program heap in the linking process and not a snapshot. The scripts `platforms/stm/bin/build.sh` and `platforms/stm/bin/link.sh` need to be modified to use the 32-bit `dartino-flashify` to generate an assembly file and compile that to an object-file to link with.

We could support both snapshot and program heap, but I don't see the need for that right now.

commit 1a30a29e (2016-03-14): Make dartino build/flash work with embedded program heap

In 49a0acf the FreeRTOS embedding started running from program heaps
instead of snapshots. This brings the Dartino SDK tooling on par with
that change.

Closes #454

BUG= https://github.com/dartino/sdk/issues/454
R=karlklose@google.com

Review URL: https://codereview.chromium.org/1799693002 .
closed 2016-03-14 by sgjesse