dart-archive/sdk #456: Support 'dartino analyze' action

mit-mit, 2016-03-18:
For regular Dart code you can verify it's correctness using the dart analyzer in the command line:

```
mit-macbookpro:samples mit$ dartanalyzer general/hello.dart
Analyzing [general/hello.dart]...
```

It would be nice to have the same thing for dartino, only integrated into our dartino command:
```
mit-macbookpro:samples mit$ dartino analyze general/hello.dart
Analyzing [general/hello.dart]...
```

mit-mit, 2016-03-18:
Discussed with @lukechurch and he OK'ed using `dartino analyze`

eernstg, 2016-03-18:
Hej Michael,

der var lige et punkt hvor jeg har en reaktion fra dybt nede i rygraden:
Det er ikke 'verify its correctness' men derimod noget klart mindre, fx
'verify its type correctness' or 'rule out certain kinds of errors' eller
noget i den stil, men der er stadig masser af rum til bugs. ;-)

mit-mit, 2016-03-18:
Yup, I was being a little colloqial

lukechurch, 2016-03-18:
@eernstg say what? :)

eernstg, 2016-03-18:
Ah, I just quipped that you can have type correct programs with bugs.

lukechurch, 2016-03-18:
We should work on fixing that - it seems very doable.

eernstg, 2016-03-18:
'ts called Coq. ;)

closed 2016-05-02 by danrubel