dart-archive/sdk #465: Running tools/test.py script remotely

StefanaCerovina, 2016-04-06:
Is it possible to run tests (tools/test.py script) using remote approach (dartino-vm on arm board and compiler on x86_64 host)? I can run individual tests or run dart program remotely using steps from: https://github.com/dartino/sdk/wiki. But how about running this python script?

midi986, 2016-05-16:
I am interested in this as well. Is it possible to run tools/test.py script remotely? In a way described above?

sgjesse, 2016-05-17:
@StefanaCerovina  sorry for the long response, and @midi986 thanks for pinging this issue.

On our [build-bots](https://build.chromium.org/p/client.fletch/console) we do exactly this. The build is produced by cross-compiling on a Linux machine, and the the test script is run on an ARM machine.

For the debug tests the bots run the following command after the cross-compiled binaries have been copied over:

```
tools/test.py -mdebug -axarm --time --report -pbuildbot --step_name=test_DebugXARM --kill-persistent-process=0 --run-gclient-hooks=0 --build-before-testing=0 --host-checked -slinux
```

The cross-buils is copied over using Google Cloud Storage, where we just create a tar-file from the `out`directory.

The buildbot script is in `tools/bots/dartino.py`. The build part of this is in `StepsCrossBuilder` and the test part is in `StepsTargetRunner`.

midi986, 2016-05-17:
Hello Soren,

Thank you for your reply. I am not sure that this is exactly what Stefana asked. Because, if I am right, instructions above are related to running both dartino-vm and compiler on arm board, not separately on arm (dartino-vm) and host PC (compiler). 

What I would like to do is to run tools/test.py script using dartino-vm on arm and compiler on host PC. The approach (separated dartino-vm and compiler on a board and host PC) described below is about to run individual dart files, but how about running all tests using tools/test.py, is it possible:

Compiling to a remote VM

Normally, dartino run will launch the dartino-vm binary. But if you want to run the dartino-vm on a device (a phone, tablet, smart lightbulb, etc.) it is often easier to start the VM and tell the compiler to attach to the already running VM. For example, on the device, run:

$ ./out/ReleaseIA32Clang/dartino-vm
Waiting for compiler on 127.0.0.1:64745

Then forward a port from your laptop to port 64745 on the device, for example, using ssh from the laptop:

$ ssh -L 64745:127.0.0.1:64745 hostname-of-device

Finally, on the laptop, run the compiler:

$ ./out/ReleaseIA32Clang/dartino attach tcp_socket 127.0.0.1:64745
Connecting to 127.0.0.1:64745
$ ./out/ReleaseIA32Clang/dartino run hello.dart

sgjesse, 2016-05-17:
The original question by @StefanaCerovina war about running tests on a remote machine, which I think is addressed by the answer above.

With regard to your question @midi986 the Dartino Agent might be what you are looking for. The Dartino Agent is a Dartino application which can run on a remote device, and can basically be used to spawn new VMs, and communicate back the port. The code for the Dartino Agent is in `pkg/dartino_agent`. This includes both the Dartino Agent code (in `bin`) and the client used by the Dartino compiler (in `lib`).

For running on the Raspberry Pi we create a .deb installation package for installing the Dartino Agent, see `tools/linux_dist_support/debian`.

In the Dartino compiler there is some magic so that if you run `dartino run test.dart in session remote` then it will look for Dartino devices using mDNS (the Raspberry Pi .deb installation package will set that up), and you can select one, or provide one manually. The choice will be remembered for "session remote".

mit-mit, 2016-05-19:
Tentatively closing this. If this still needs to be addressed, please reopen with more details.

closed 2016-05-19 by mit-mit