dart-archive/sdk #466: type 'UnixDomainAddress' is not loaded

StefanaCerovina, 2016-04-06:
Unhandled exception:
'package:fletchc/src/hub/hub_main.dart': malformed type: line 229 pos 40: type 'UnixDomainAddress' is not loaded
server = await ServerSocket.bind(new UnixDomainAddress(socketFile.path), 0);
^

#0 _TypeError._throwNew (dart:core-patch/errors_patch.dart:56)
#1 main. (package:fletchc/src/hub/hub_main.dart:229:40)
#2 Future.Future.microtask. (dart:async/future.dart:144)
#3 _microtaskLoop (dart:async/schedule_microtask.dart:41)
#4 _startMicrotaskLoop (dart:async/schedule_microtask.dart:50)
#5 _runPendingImmediateCallback (dart:isolate-patch/isolate_patch.dart:96)
#6 _RawReceivePortImpl._handleMessage (dart:isolate-patch/isolate_patch.dart:149)
./fletch: Failed to start fletch server (Connection refused).
Use DART_VM environment variable to override location of Dart VM.

I have tried to build dartino on arm board, and use manually cross compiled dart on x64 machine.

How can I build dart executable for arm platform, and use it for dartino project?

sgjesse, 2016-04-06:
Looks as if you are using a quite old checkout. The dependency on Unix Domain Sockets (using a forked Dart VM) was removed in https://github.com/dartino/sdk/commit/0c534762a7d12b9d9cc69cb58a5d03a06da16db2.

Also the project was renamed from Fletch to Dartino in https://github.com/dartino/sdk/commit/4c86cb1724a704917a0916b1457495d37b1990d1, where `package:fletchc` was renamed to `package:dartino_compiler`.

closed 2016-04-06 by sgjesse

sgjesse, 2016-04-06:
Regarding your second question on a dart executable for ARM, then we do build a Dart SDK for ARM, but we don't distribute it at the moment. If you look at the bottom of https://www.dartlang.org/downloads/archive/ you can use `arm` as the architecture to get a Dart SDK for ARM, e.g.

```
$ curl -o dartsdk-linux-arm-release.zip https://storage.googleapis.com/dart-archive/channels/stable/release/1.15.0/sdk/dartsdk-linux-arm-release.zip
```

@whesse can you give more details on the actual ARM architecture? It is most likely `armv7`.

whesse, 2016-04-06:
Yes, this is an armv7 build.  We have target options to build armv6 and armv5 builds of the Dart vm, but the default one we build just has cflags:'-marm'  '-mfpu=vfp' and is built with gcc-cross/arm-linux-gnueabihf/4.8.4

If you run arm-linux-gnueabihf-gcc -v on this version, it prints its default options, including
--with-arch=armv7-a --with-fpu=vfpv3-d16 --with-float=hard --with-mode=thumb