dart-archive/sdk #469: Dartino fails in Bash on Windows

mit-mit, 2016-04-11:
Just tried running dartino in the new Bash on Windows shell, and it fails with a file descriptor issue. 

It would be great if we could create a smaller (more self contained) repro, and log an issue here:
https://github.com/Microsoft/BashOnWindows/issues

Repro steps:

1. Install the last insider build of Windows 10
1. Open a 'Bash on Ubuntu on Windows' window
1. Run the following commands:
```
apt-get install unzip

wget https://storage.googleapis.com/dartino-archive/channels/dev/release/latest/sdk/dartino-sdk-linux-x64-release.zip

unzip dartino-sdk-linux-x64-release.zip

./dartino-sdk/bin/dartino --version
```

Expected result: The version number is printed
Actual result:
```
root@localhost:~# ./dev/dartino-sdk/bin/dartino --version
../../src/tools/driver/platform_linux.cc:73: error: signalfd failed: Invalid argument
Aborted (core dumped)
```

mit-mit, 2016-04-11:
@sgjesse unfortunately this did not work

sgjesse, 2016-04-11:
This is the code that fails

```
  int fd = signalfd(-1, &signal_mask, SFD_CLOEXEC);
  if (fd == -1) {
    FATAL1("signalfd failed: %s", strerror(errno));
  }
```

Maybe some of the signals we in the `signal_mask` is not supported on "Windows".

Is there a description of what could be "API limitations"?

mit-mit, 2016-04-11:
Posted a question here: https://github.com/Microsoft/BashOnWindows/issues/129

sgjesse, 2016-04-13:
Copy of the answer from Microsoft

Thanks for trying this out. 'signalfd' syscall is not yet implemented in WSL. No ETA at this time. We are looking at getting the (un)implemented syscall list documented somewhere. Stay tuned!

closed 2016-07-19 by mit-mit