dart-archive/sdk #470: Error: no agent: SocketException: OS Error: Operation timed out, errno = 60, address = 192.168.2.2, port = 51291

filiph, 2016-04-11:
Can't connect to a newly flashed Rasp Pi 2 (that did work with Fletch).

Connecting from Mac OS X through USB-Ethernet cable. At first I tried automatic DHCP but that just led to self-assigned IP on my workstation. I re-flashed the SD card with a static IP (192.168.2.2).

Now, when I run `dartino show devices`, I get this:

```
$ dartino show devices
Using template settings file '/Applications/dartino-sdk/internal/.dartino-settings'
Looking for Dartino capable devices (will search for 5 seconds)...
Device at 192.168.2.2      dartinopi.local (Error: no agent: SocketException: OS Error: Operation timed out, errno = 60, address = 192.168.2.2, port = 51291)
```

Dartino version: 0.3.0-dev.5.2
Dart VM version: 1.15.0 (Wed Mar  9 09:55:45 2016) on "macos_x64"
Mac OS X 10.11.4

I'll try rebooting now but I wanted to report before I lose track of this.

filiph, 2016-04-12:
Update: Rebooting didn't help. Nor did connecting the Raspberry Pi to the internet (for clock sync).

I did some research and looked into `/etc/init.d/dartino-agent`. Running `/etc/init.d/dartino-agent status` tells me the agent is `active (running)`.

filiph, 2016-04-13:
Ok, I'm a Linux networking noob, so pardon my cluelessness. I solved/sidestepped the issue by:

* Connecting the Rasp Pi to a monitor and keyboard (took me a day to get the correct video cable...)
* Going to terminal
* running `hostname -I`

This resulted in two IP addresses, something like this:

```
$ hotname -I
192.168.1.123 169.254.64.68
```

`192.168.1.123` is the one I entered when running `platforms/raspberry-pi2/flash-sd-card`. But it's the second one that worked in the end.

`dartino show devices` still times out, but `dartino run path/to/script.dart in session remote` works (after I deleted the previous `remote.dartino-settings` and entered the second IP address).

sgjesse, 2016-04-13:
This looks as if the routing from the Mac to the 192.168.1.123 address does not go to the USB adapter.

@filiph How did you configure the USB adapter on the Mac? You you should have a manual IP address of e.g. 192.168.1.1 and netmask of 255.255.255.0 for that adapter.

@karlklose can you take a look? Maybe our error reporting could be better here. I guess that the MDNS broadcast gets through on all adapters regardless of the routing.

Does the RPi always get a selfassigned 169.254.0.0/16 address when we configure the IP address in `cmdline.txt` on the boot partition?

karlklose, 2016-04-13:
Filip, could you run ifconfig on both you Mac and the Raspberry Pi and send
the output to me?

gruberjuliannama, 2016-04-13:
Operation timed out, errno = 60, address = 192.168.2.2, port = 51291 to Error: no agent: SocketException: OS Error

filiph, 2016-04-13:
@sgjesse Hmm. I went with "Configure IPv4: Using DHCP with manual address" which, in retrospect, was probably wrong. It gives me the unchangeable netmask of 255.255.255.255.

I went with manual, changed the netmask to 255.255.255.0 and — **it works!** `dartino show devices` shows dartinopi just fine.

Thanks for the help.

When I have more time, I'll look into maybe giving some hints for clueless users like me in the getting started guide.

@karlklose In case you still need it, here are the ifconfig outputs. For **Mac**:
```
$ ifconfig
lo0: flags=8049<UP,LOOPBACK,RUNNING,MULTICAST> mtu 16384
	options=3<RXCSUM,TXCSUM>
	inet6 ::1 prefixlen 128
	inet 127.0.0.1 netmask 0xff000000
	inet6 fe80::1%lo0 prefixlen 64 scopeid 0x1
	nd6 options=1<PERFORMNUD>
gif0: flags=8010<POINTOPOINT,MULTICAST> mtu 1280
stf0: flags=0<> mtu 1280
en0: flags=8863<UP,BROADCAST,SMART,RUNNING,SIMPLEX,MULTICAST> mtu 1500
	ether 6c:40:08:99:41:3e
	inet6 fe80::6e40:8ff:fe99:413e%en0 prefixlen 64 scopeid 0x4
	inet6 2620::1000:5f44:6e40:8ff:fe99:413e prefixlen 64 autoconf
	inet6 2620::1000:5f44:2818:eb4:728a:dd7b prefixlen 64 autoconf temporary
	inet 172.17.193.184 netmask 0xfffffe00 broadcast 172.17.193.255
	nd6 options=1<PERFORMNUD>
	media: autoselect
	status: active
en1: flags=963<UP,BROADCAST,SMART,RUNNING,PROMISC,SIMPLEX> mtu 1500
	options=60<TSO4,TSO6>
	ether 72:00:05:f9:6e:70
	media: autoselect <full-duplex>
	status: inactive
en2: flags=963<UP,BROADCAST,SMART,RUNNING,PROMISC,SIMPLEX> mtu 1500
	options=60<TSO4,TSO6>
	ether 72:00:05:f9:6e:71
	media: autoselect <full-duplex>
	status: inactive
p2p0: flags=8843<UP,BROADCAST,RUNNING,SIMPLEX,MULTICAST> mtu 2304
	ether 0e:40:08:99:41:3e
	media: autoselect
	status: inactive
awdl0: flags=8943<UP,BROADCAST,RUNNING,PROMISC,SIMPLEX,MULTICAST> mtu 1484
	ether 9a:6b:d7:cb:0b:ee
	inet6 fe80::986b:d7ff:fecb:bee%awdl0 prefixlen 64 scopeid 0xa
	nd6 options=1<PERFORMNUD>
	media: autoselect
	status: active
bridge0: flags=8863<UP,BROADCAST,SMART,RUNNING,SIMPLEX,MULTICAST> mtu 1500
	options=63<RXCSUM,TXCSUM,TSO4,TSO6>
	ether 6e:40:08:99:2a:00
	Configuration:
		id 0:0:0:0:0:0 priority 0 hellotime 0 fwddelay 0
		maxage 0 holdcnt 0 proto stp maxaddr 100 timeout 1200
		root id 0:0:0:0:0:0 priority 0 ifcost 0 port 0
		ipfilter disabled flags 0x2
	member: en1 flags=3<LEARNING,DISCOVER>
	        ifmaxaddr 0 port 5 priority 0 path cost 0
	member: en2 flags=3<LEARNING,DISCOVER>
	        ifmaxaddr 0 port 6 priority 0 path cost 0
	nd6 options=1<PERFORMNUD>
	media: <unknown type>
	status: inactive
en3: flags=8863<UP,BROADCAST,SMART,RUNNING,SIMPLEX,MULTICAST> mtu 1500
	options=4<VLAN_MTU>
	ether 10:9a:dd:42:3f:66
	inet6 fe80::129a:ddff:fe42:3f66%en3 prefixlen 64 scopeid 0xc
	inet 192.168.1.1 netmask 0xffffffff broadcast 192.168.1.1
	nd6 options=1<PERFORMNUD>
	media: autoselect (100baseTX <full-duplex,flow-control>)
	status: active
pflog0: flags=1<UP> mtu 33080
pflog1: flags=1<UP> mtu 33080
```

And here for **dartinopi**:
```
$ ifconfig
eth0      Link encap:Ethernet  HWaddr b8:27:eb:e3:99:c9
          inet addr:192.168.1.123  Bcast:192.168.1.255  Mask:255.255.255.0
          inet6 addr: fe80::ba27:ebff:fee3:99c9/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:134 errors:0 dropped:0 overruns:0 frame:0
          TX packets:139 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:17015 (16.6 KiB)  TX bytes:24095 (23.5 KiB)

lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:260 errors:0 dropped:0 overruns:0 frame:0
          TX packets:260 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:21072 (20.5 KiB)  TX bytes:21072 (20.5 KiB)
```

karlklose, 2016-04-20:
I think we should improve the instructions to make sure that people set the correct netmask (by using "Manually" on MacOS).
The timeout problem is already tracked in #458, so I am closing this issue.

closed 2016-04-20 by karlklose