dart-archive/sdk #481: Dartino analyze fails when run from a repo

mit-mit, 2016-05-03:
Synced, pulled, and rebuilt, and now:
```
./out/ReleaseX64/dartino analyze samples/general/hello.dart
```

gives me:
```
Using template settings file '/Users/mit/dev/github/dartino/sdk/.dartino-settings'
The Dartino program is broken and has crashed.

The Dartino team would greatly appreciate if you would take a moment to
report this problem at https://github.com/dartino/sdk/issues/new

Please include the following information:

* the name and version of your operating system

* the Dartino SDK version (0.4.0-edge.e2d5fd6f19c6df733927dcf15cb2c7ddde0c9e47)

* the entire message you see here (including the full stack trace below)

'package:dartino_compiler/src/worker/developer.dart': Failed assertion: line 1306 pos 12: 'await dartSdkDirectory.exists()' is not true.
#0      _AssertionError._throwNew (dart:core-patch/errors_patch.dart:27)
#1      locateDartSdkDirectory.<locateDartSdkDirectory_async_body> (package:dartino_compiler/src/worker/developer.dart:1306:12)
#2      _asyncThenWrapperHelper.<anonymous closure> (dart:async-patch/async_patch.dart:27)
#3      _rootRunUnary (dart:async/zone.dart:902)
#4      _CustomZone.runUnary (dart:async/zone.dart:798)
#5      _Future._propagateToListeners.handleValueCallback (dart:async/future_impl.dart:551)
#6      _Future._propagateToListeners (dart:async/future_impl.dart:637)
#7      _Future._completeWithValue (dart:async/future_impl.dart:424)
#8      _Future._asyncComplete.<anonymous closure> (dart:async/future_impl.dart:479)
#9      _rootRun (dart:async/zone.dart:895)
#10     _CustomZone.run (dart:async/zone.dart:790)
#11     _CustomZone.bindCallback.<anonymous closure> (dart:async/zone.dart:723)
#12     _microtaskLoop (dart:async/schedule_microtask.dart:41)
#13     _startMicrotaskLoop (dart:async/schedule_microtask.dart:50)
#14     _runPendingImmediateCallback (dart:isolate-patch/isolate_patch.dart:96)
#15     _RawReceivePortImpl._handleMessage (dart:isolate-patch/isolate_patch.dart:149)

Session 'local' was terminated.
```

danrubel, 2016-05-04:
If you have built the sdk, then wouldn't you run `./out/ReleaseX64/dartino-sdk/bin/dartino analyze samples/general/hello.dart` ?

danrubel, 2016-05-04:
Talked with Soren this morning and now understand what needs to be done and why. Will fix. Sorry for the noise.

mit-mit, 2016-06-29:
Bumping to 0.6.0

closed 2016-06-30 by danrubel