dart-archive/sdk #482: Include foreign binary

mit-mit, 2016-05-03:
FFI can already be used to call into native from Dartino (and to get callbacks back into Dartino). But we need to support including the binary library in the image that is build when the developer runs `dartino compile` or `dartino flash`.

Perhaps they should list the binaries in the `dartino.yaml` file?

mit-mit, 2016-05-03:
Tentatively adding this to 0.4.0 to see if there is some initial version of this we can do?

mit-mit, 2016-05-09:
We discussed the following approach:

Assume a developer wants to use three methods from two libraries:

Methods `method1` and `method2` located in library `foo` available as `libfoo.so/libfoo.a/libfoo.h`.
Method `other_method_3` located in library `bar` available as `libbar.so/libbar.a/libbar.h`.

The developer would take the following steps for a Dartino application located in `[appdir]`:

- Copy `libfoo.so`, `libfoo.a`, `libfoo.h`, `libbar.so`, `libbar.a`, and `libbar.h` to `[appdir]/ffi`
- Create a new file `ffisymbols.c`in `[appdir]/ffi` with the following contents:

```
DARTINO_EXPORT_STATIC_RENAME(method1, Method1)
DARTINO_EXPORT_STATIC_RENAME(method2, Method2)
DARTINO_EXPORT_STATIC_RENAME(other_method_3, Method3)
```

With that, the binaries will be included in the linked image.

To actually use the methods from a Dartino program, the developer can then add in their `.dart` file:

- To make sure the `.so` file can be dynamically loaded when running locally on the PC:
```
final ForeignLibrary libfoo = new ForeignLibrary.fromName(ForeignLibrary.bundleLibraryName('foo'));
final ForeignLibrary libbar = new ForeignLibrary.fromName(ForeignLibrary.bundleLibraryName('bar'));
```

- To use the methods:

```
final method1= libfoo.lookup('Method1');
final method2= libfoo.lookup('Method2');
final method3= libbar.lookup('Method3');
```

ricowind, 2016-05-09:
Maybe add a small helper to the FFI package:
```
ForeignLibrary loadLibraryAgnostic(String libName) {
  return (Foreign.platform == Foreign.FREERTOS)
      ? ForeignLibrary.main 
      : new ForeignLibrary.fromName(ForeignLibrary.bundleLibraryName(libName));
}
```
To avoid having people do all the boilerplate (and to easily add more board support without having people change their code :-)).

commit 39ff5422 (2016-05-13): Optionally include foreign libraries when building embedded deployment

This adds a simple way of including additional libraries (.a or .o
files) when running 'dartino build' and 'dartino flash'. All files in
the ffi subdirectory relative to the .dart file being compiled are added
to the link step.

Added a sample of how to compile a native library and an additional
static ffi symbol file.

Also added an intrinsic set of static ffi functions (currently only
strlen) because the ffi package use C function from the main library.

This needs to be further refined.

Closes #482

BUG= https://github [... cut]
commit 11a2c68b (2016-05-13): Optionally include foreign libraries when building embedded deployment

This adds a simple way of including additional libraries (.a or .o
files) when running 'dartino build' and 'dartino flash'. All files in
the ffi subdirectory relative to the .dart file being compiled are added
to the link step.

Added a sample of how to compile a native library and an additional
static ffi symbol file.

Also added an intrinsic set of static ffi functions (currently only
strlen) because the ffi package use C function from the main library.

This needs to be further refined.

Closes #482

BUG= https://github [... cut]
commit d3f85427 (2016-05-13): Optionally include foreign libraries when building embedded deployment

This adds a simple way of including additional libraries (.a or .o
files) when running 'dartino build' and 'dartino flash'. All files in
the ffi subdirectory relative to the .dart file being compiled are added
to the link step.

Added a sample of how to compile a native library and an additional
static ffi symbol file.

Also added an intrinsic set of static ffi functions (currently only
strlen) because the ffi package use C function from the main library.

This needs to be further refined.

Closes #482

BUG= https://github [... cut]
closed 2016-05-13 by sgjesse