dart-archive/sdk #483: Process support does not work on STM board

mit-mit, 2016-05-09:
Process support appears to now work properly on the STM F7 board. 

Repro instuctions:

- Download and extract the attached .zip file
[repro.zip](https://github.com/dartino/sdk/files/254781/repro.zip)

- Run on the local PC: `$ ~/tmp/dartino-sdk/bin/dartino run ./repro2.dart`
The output is properly interleaved:

```
Fibonacci of 31 is 1346269
Fibonacci of 32 is 2178309
Fibonacci of 33 is 3524578
I am still alive 2
Fibonacci of 34 is 5702887
Fibonacci of 35 is 9227465
I am still alive 3
Fibonacci of 36 is 14930352
I am still alive 4
I am still alive 5
Fibonacci of 37 is 24157817
I am still alive 6
I am still alive 7
Fibonacci of 38 is 39088169
I am still alive 8
I am still alive 9
I am still alive 10
```

- Then run on the board: `$ ~/tmp/dartino-sdk/bin/dartino flash ./repro3.dart`. Notice how the output is not interleaved -- the counter is only updated every time a new fib number is ready.

mit-mit, 2016-05-27:
@sgjesse should this be assigned to you, someone else?

sgjesse, 2016-05-27:
I have it on my list.

mit-mit, 2016-06-29:
Resolved in https://codereview.chromium.org/2105833002/

closed 2016-06-29 by mit-mit